winder/Universal-G-Code-Sender #727: Parsing issues

groberts22, 2017-08-30:
I am using the 8/30 nightly build, and continue to have problems with the parser. The screenshot below shows a typical example.  Something somewhat similar happens when I issue G0 commands on 2 consecutive lines with references to the same axis (i.e., if I combine lines 12 and 13 below into a single command, I get an error telling me I have multiple G codes requiring parameter references in the same block).

[image]

winder, 2017-08-30:
Do you have anything in the "Regular Expression Pattern Remover" configuration?

http://winder.github.io/ugs_website/guide/troubleshooting/#gcode-program-stopped-working

[image]

groberts22, 2017-08-31:
I checked and there is nothing in the "Regular Expression Pattern Remover" configuration. I also did some further testing.  The script appears to run fine when I send it to the machine.  It is only when I attempt to test the code using the $C option that I get the errors.  Usually it is either error code 20 or error code 1. (see below) The UGS program then hangs up and I have to disconnect and reconnect to the router to get the UGS program to respond again. The problem does not always occur at the same place in the code, and (rarely, perhaps once or twice) the script has completed successfully in test mode.  

Example:
(error:20) Unsupported or invalid g-code command found in block.
**** Pausing file transfer. ****

(error:20) Unsupported or invalid g-code command found in block.
(error:1) G-code words consist of a letter and a value. Letter was not found.
**** Pausing file transfer. ****

(error:1) G-code words consist of a letter and a value. Letter was not found.
ok

winder, 2017-09-11:
Could you attach or email me a file to reproduce the problem?

jahnj0584, 2017-09-11:
E-mailed to wwinder.unh@gmail.com

vMeph, 2017-09-11:
Click winder picture it will take you to his porfile
Email is there

winder, 2017-09-19:
@chamnit could this error be related to sending grbl commands while it is processing the G10 and updating the EEPROM? I was reading through the code and this part caught my eye in particular:
```
// However, this doesn't prevent issues with lost serial RX data during an EEPROM write, especially
// if a GUI is premptively filling up the serial RX buffer simultaneously. It's highly advised for
// GUIs to flag these gcodes (G10,G28.1,G30.1) to always wait for an 'ok' after a block containing
// one of these commands before sending more data to eliminate this issue.
```

I have a single step mode available, but I don't turn it on/off for specific commands. Maybe I need to for users with custom probing scripts like this?

winder, 2017-09-19:
Oops, I see it in the documentation now:

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#eeprom-issues

chamnit, 2017-09-19:
Yes. This is what happens when you try to continually send commands to Grbl while its writing to EEPROM. In this case, G10.

winder, 2017-09-19:
@groberts22 @BusbyGT adding `G4 P0.01` after the `G10` command should fix this until I can get a more permanent fix in place.

chamnit, 2017-09-19:
@winder : A `G4` command won't help. The problem is the EEPROM write shuts down the serial port, so the G4 command will be truncated or lost completely before Grbl reads it.

FWIW, most people set their work coordinates ad hoc. It's very rare that someone tries to do it in g-code. If they do, it's typically not part of a program, but usually a quick task to set up the job, like this user is doing with probing. They just need to run this in simple streaming mode until you can make the adjustment.

BusbyGT, 2017-09-19:
For me, that script has previously worked, it just was broken starting around the time the Gcode parser broke for G38.2 commands. I'm not having an issue with the G10, I'm having an issue with the G38 not running after a G0 command. While the EEPROM issue is a concern, I don't think that's why my issue is happening.

chamnit, 2017-09-19:
@BusbyGT : If the EEPROM write is the issue, then the G10 would work fine. The following commands like you G38 and G0 would likely get truncated. See the Wiki for details on why there is such a problem. Maybe it'll become more clear.

BusbyGT, 2017-09-19:
@chamnit I understand what you're saying. My wording may not have been the best.

My issue was that UGS is not parsing my input file completely. Back in June, before UGS enhanced the Gcode parser, this worked flawlessly when UGS was loading my entire file. Now, I don't even think my third G38.2 even sends to GRBL from UGS. It's possible that this might have been a race condition and now other changes in UGS made it start showing up. I'll try to do some more thorough investigation using UGS Classic since that gives me error popups. I believe the EEPROM issue is one that needs to be considered and accounted for but may not be my issue. I'll revisit my other issue report if I find out the issue is UGS not loading the script as opposed to a GRBL issue.

winder, 2017-09-20:
@BusbyGT I was able to fix this problem and test it using the advanced probing widget I'm working on -- last night I noticed this problem and put a 250ms delay after the `G10` as a temporary workaround. With the fix I no longer need that delay. I'm in the middle of a lot of other changes right now so I wont be able to get the changes into the nightly build for a bit.

If you have a java compiler, the change is in `GrblCommunicator.java`. I'll get it into the nightly build ASAP.
```java
    private boolean temporarySingleStepMode;
    private final static String EEPROM_COMMAND_PATTERN = "G10|G28|G30|\\$x=|\\$I|\\$N|\\$RST=|G5[456789]|\\$\\$|\\$#";
    private final static Pattern EEPROM_COMMAND = Pattern.compile(EEPROM_COMMAND_PATTERN, Pattern.CASE_INSENSITIVE);

    /**
     * When a command is sent, check if it is one of the special commands which writes to the EEPROM.
     * If it is temporarily setSingleStepMode(true) to avoid corruption.
     */
    @Override
    protected void sendingCommand(String response) {
        // If this is an EEPROM command switch to single step mode temporarily.
        if (EEPROM_COMMAND.matcher(response).find()) {
            this.temporarySingleStepMode = !this.getSingleStepMode() || this.temporarySingleStepMode;
            this.setSingleStepMode(true);
        } else if (this.temporarySingleStepMode) {
            this.temporarySingleStepMode = false;
            this.setSingleStepMode(false);
        }
    }
```

BusbyGT, 2017-09-20:
As I mentioned in #769, my issue is not related to this, right now. It was a Gcode parsing error in UGS that wasn't loading the probe script fully and not sending the last G38.2 file. Now that I know the issue, I'll correct my scripts and try it again next time I use the machine.

chamnit, 2017-09-20:
@BusbyGT : Again, when EEPROM writes, the serial port disables inside Grbl. UGS may be sending the G38.2 commands properly, but they are not getting fetched from the serial RX hardware register when they are received. By the time the EEPROM write is complete, one or two characters may have been sent to Grbl, been overwritten, and lost. That's what I mean by truncated. 

The type of error you were receiving is exactly what happens when this EEPROM problem occurs and is not handled properly by the GUI streamer. One character was lost from a line and Grbl reports back an error since that character mangled the command just enough to throw an error. 

The EEPROM issue is not something that can be fixed and intrinsic to how an AVR processor works. The issue is well-documented and there is a suggested workaround for GUIs that use the character counting protocol. Once Will fixes the problem, which it looks like he has, this problem will go away.

winder, 2017-09-20:
@BusbyGT you're just finding all the rough edges in the UGS parser right now, we should get them figured out pretty soon. I should have been more careful with the recent parser changes, having multiple variables made this harder to troubleshoot than it should have been.

As always challenges like this just make UGS better. The parser has a bunch of new unit tests to help guard against these sorts of regressions in the future and the more accurate parser should enable more advanced transformations -- things like pan/rotate/crop/split.

It's particularly nice to finally have a fix for the EEPROM issue, a side effect is that you should be able to store your settings in a gcode file and stream them for a backup/restore type operation. Restore was such a rare operation that I never bothered, but these probe scripts seem a lot more common so it finally got onto my TODO list.

winder, 2017-09-28:
@BusbyGT @groberts22  these changes are now in the nightly build. Your original file with the extra space should work now. The code to automatically switch to the send-response protocol for commands that use EEPROM is also enabled.

closed 2017-09-28 by winder