winder/Universal-G-Code-Sender #729: Spindle Dwell?

jahnj0584, 2017-09-04:
Any chance we can add a user configurable spindle dwell time for when M3 commands are initiated? Very few, if any VFDs are able to get to speed immediately. It seems like Mach, UCCNC, and linuxCNC all have this feature already too.

winder, 2017-09-04:
It should be possible to throw in a `G4 PXXX` after detecting an `M3`. I think GRBL might already do this though based on this comment: https://github.com/gnea/grbl/issues/28#issuecomment-257126519

jahnj0584, 2017-09-04:
> think GRBL might already do this though based on this comment: gnea/grbl#28

Who do I ask to get that added?

winder, 2017-10-14:
I decided to build this feature as part of a [tutorial for creating a new gcode processor](http://winder.github.io/ugs_website/dev/gcode_processor/#tutorial-creating-a-new-commandprocessor).

[image]

closed 2017-10-14 by winder

jahnj0584, 2017-10-25:
coding is over my head, how about you just add it to UGS from the get go? :P

I had a friend edit my post for me, to insert G04 P0.5 after an M3

winder, 2017-10-25:
There's no coding involved, it's a new option on the gcode parser config
page

commit edc9f08e (2017-10-14): Add M3-Dweller processor to add G4 command following M3 spindle enable commands. Closes #729