winder/Universal-G-Code-Sender #730: Spindle over ride doesn't work

jahnj0584, 2017-09-04:
Just got my VFD wired up. M3 S--- works and so does M5. Pressing +/-/reset in the over rides window has 0 effect on my spindle. It only sends a 9e3x or something command in the console.

winder, 2017-09-05:
Does the percentage in the GUI change? That is the override value as GRBL reports it, so if it changes then there is something going on in the wiring/grbl/grbl-configuration.

jahnj0584, 2017-09-05:
Yes, it displays the new % value but I still get 4.7v out of the PWM no
matter what it's set to (5v is the arduino Max on paper)

It WILL change the speed if I send a new M3 S--- command so the PWM
function is working fine.

winder, 2017-09-05:
I'd ask on the grbl project then, seems the override is being set properly. Could be a config issue, a bug, or maybe the features just doesn't work the way you expect it to.

closed 2017-09-05 by winder

vMeph, 2017-09-08:
@jahnj0584  would be nice if you could provide info on wiki on how to wire VFD systems  for others have a start point

jahnj0584, 2017-09-11:
Which wiki? Could channit add it to the grbl one?