winder/Universal-G-Code-Sender #732: 2.0 build automation

winder, 2017-09-04:
After #731 is done the builds need to be automated. Unfortunately, Jenkins may not be suitable for this anymore since the `Mac OS` build may not work properly using Linux because of differences between `hdiutil` (mac only) and the Linux alternative `genisoimage`.

The artifacts should also be deployed to Sourceforge instead of depending on cloudbees.

One option for a Jenkins alternative is TravisCI, which has mac build machines available.

winder, 2018-01-16:
I have an automated build for Windows/Linux which includes Java:
[Linux](https://winder.ci.cloudbees.com/job/UGS-Installers/lastSuccessfulBuild/artifact/UGSPlatform/dist/ugsplatform-installer-linux.sh
)
[Windows](https://winder.ci.cloudbees.com/job/UGS-Installers/lastSuccessfulBuild/artifact/UGSPlatform/dist/ugsplatform-installer-windows.exe)

There's still some work to do to get the OSX installer working. I can run it manually, but the automated build machine runs Linux and one of the utilities I need is missing.

winder, 2018-01-16:
Please give the above links a try and let me know if they work. I was able to install them on 32/64 bit Windows/Linux virtual machines. If you try one please include the OS and architecture you're running on.

breiler, 2018-03-27:
I'm curious about the problems you have with MacOS-builds. Why doesn't the image from genisoimage work?

winder, 2018-03-27:
@breiler the utility isn't available on the jenkins machine. I already include some [native 7zip binaries](https://github.com/winder/Universal-G-Code-Sender/tree/master/ugs-platform/application/installer/7zip), so should probably just include it in the build.

I made a blog post [about the OSX bundle](http://blog.willwinder.com/2017/09/netbeans-platform-mac-osx-app-bundle.html) here for the full story about this implementation since it was pretty involved. I should make another for the Linux/Windows/RPi bundles as well since they are pretty involved too and do some novel stuff around creating the JRE bundle.

breiler, 2018-03-27:
I can give it a try. You have probably already checked, does the Jenkins machine have `mkisofs`?

winder, 2018-03-27:
I'm not sure, I've had a hard time getting information out of cloudbees. If you create an account at https://www.cloudbees.com/ I can invite you to the project.

breiler, 2018-04-03:
I've done som experimentation and would like some input.

I got the installer to work on MacOSX, but it still need some work. The generated executable script created by Netbeans is missing the executable flag. If the user have java 9 the installer doesn't work at all. The process of creating the installer is not super clear and I think it is too complex... As you wrote in your blog it's not the expected way of installing programs on some operating systems. I think we should scrap the installer (at least for MacOSX).

The other option I wanted to try was with the dmg-images with Travis, which provides the means to install additional software required. It doesn't have a plugin for uploading binaries to SF, so I tried Bintray and Github. Bintray has the upload limit of 50Mb per file. Github seems to work and doesn't have any download quota limits (that I could find).

I tried setting up a build process for nightly builds and releases with Github and still have some problems with triggering and tagging from Travis. Before I continue I would like to know if this is a path worth exploring?

The results from a build would look like this:
https://github.com/breiler/Universal-G-Code-Sender/releases

Example of travis.yml (WIP):
https://github.com/breiler/Universal-G-Code-Sender/blob/master/.travis.yml

winder, 2018-04-03:
When you say scrap the MacOSX installer are you referring to the one built into the Netbeans harness? I'm completely fine with that (which is why I spent time on the `create-dmg` target, though that is pretty complicated too). I don't think bundled JRE works with the built in MacOSX installer either. The `create-dmg` target could be simplified by leveraging the image buildable with `build-jre-bundles.xml`.

Moving to Travis is fine, one big benefit that it has is support for native MacOSX build nodes. The Gephi project is utilizing Travis as well:
https://github.com/gephi/gephi/blob/master/.travis.yml

I'm not opposed to hosting binaries somewhere else, the current hosting for nightly builds on cloudbee's isn't ideal. I've heard good things about artifactory/bintray but haven't looked into it, I think this is probably the right way to go, but the 1TB download limit for OSS probably isn't enough for UGS (especially once there is a bundled JRE). One wishlist item that I have is good download statistics, especially for auto-update artifacts - does the free version of bintray have this? I might consider github for milestone releases, but I think it would probably be too noisy to put nightly builds there.

breiler, 2018-04-03:
Yes, I meant scrapping the installer from Netbeans.

There is statistics in bintray, but I have no idea if it's any good. I tried downloading a couple of files and it doesn't show... The files may not be larger than 50MB when uploading from Travis, and as you said the 1TB download limit. So I feel that bintray isn't applicable for UGS.

And you are right about github, it might get bloated.

You are using sourceforge for the update site, I could try to set up a script in travis to copy binaries there instead?

winder, 2018-04-03:
It looks like Artifactory has an OSS sponsorship program, I sent in an application. If it's approved it looks like the best option. Cloudbee's and POEditor are used thanks to similar sponsorship programs with those companies.

breiler, 2018-04-04:
Sweet!

winder, 2018-05-12:
@breiler we have an Artifactory server :)
I took a stab at some basic integration and it looks like it should work nicely: https://github.com/winder/Universal-G-Code-Sender/blob/deployments/.travis.yml#L14

I need to do some more work to figure out permissions and best practices.

breiler, 2019-02-07:
How are things progressing with travis?

I tried setting up an instance and ran the travis-script in the deployments-branch and got some problems. I had to remove the deployment-profile. 

What do you think about this:
https://github.com/breiler/Universal-G-Code-Sender/blob/deployments/.travis.yml

I replaced the artifactory username and password with the API-key which feels a bit safer?
I would also love to see the jar-artifacts to be stored. That way a plugin could be written in it's own project linking to UGS-libraries.

breiler, 2019-10-07:
A nightly build for Mac is built here, needs a bit of testing though:
https://ugs.jfrog.io/ugs/webapp/#/artifacts/browse/tree/General/UGS/nightly/ugs-platform-app-2.0-SNAPSHOT.dmg

breiler, 2019-11-15:
Hi @winder,

I've made an attempt to build packages for windows, raspberry pi, and linux (x64):
https://ugs.jfrog.io/ugs/webapp/#/artifacts/browse/tree/General/UGS/nightly

The dot in the filename was not a good idea, could you remove this:
https://ugs.jfrog.io/ugs/webapp/#/artifacts/browse/tree/General/UGS/nightly/ugs-platform-app-2.0-SNAPSHOT.win.zip

winder, 2019-11-15:
very cool, thanks @breiler

breiler, 2020-01-07:
I'm closing this as the new packages are in place.

closed 2020-01-07 by breiler