winder/Universal-G-Code-Sender #735: Usability Improvements: Connection Toolbar

winder, 2017-09-04:
Perhaps specific to my use case, but is the connection bar deserving of such prominent placement in the UI? Once a connection is established it all becomes irrelevant until you're done actually using the app, and even then it's only the disconnect button certainly not the initial configuration options. The real downside of this is that you're pushing over far more relevant UI controls, the Play/Pause/Stop set. I would imagine a heat map of normal use would indicate the connection bar is a deadzone despite it's prominent placement in the global app interface.

- [] Hide the firmware/port/baud buttons after a connection is made.

jahnj0584, 2017-09-04:
Good call. Make the com port and baud a drop down option either in settings
or in the 2nd workspace where you have the editable $$ table ;). Once these
are set they probably will never change. Maybe a tool tip or pop up window
when a version is first loaded so newbies know to select it first?

Just put a dis/connect button in the corner and simply make it a rectangle.
Could also just go under common actions?

jahnj0584, 2017-09-04:
I also have an issue personally with the colors. I feel like green should
be to connect and red to close it, not the other way. I KNOW this is more
to display it's current status but it feels like Im hitting connect to end
my day

winder, 2017-09-10:
New toolbar behavior

[image]

[image]

closed 2017-09-10 by winder

jahnj0584, 2017-09-11:
Whats the controller icon? It has no title in the tools dropdown and doesnt do anything if you click it

vMeph, 2017-09-11:
Its a aplication pendant that have alot bugs ,when you click on it shows a ip, with a smartphone conected to the same wifi network, you put that ip on google adress on smartphone ,and you can control and see what machine doing from distance on the smartphone, to bad it have some bugs