winder/Universal-G-Code-Sender #736: Usability Improvements: Browse toolbar improvements

winder, 2017-09-04:
Unless you know it to be common for people to manually type file paths, the text input might as well trigger the same behavior as the adjacent browse button. As the input is a bigger target, users will naturally aim for it if functionality permits. Then again, if no one is typing into it there's no reason for the input to exist when empty. Once a file is loaded the file name is displayed there in an editable format despite that changing the value has no apparent effect. Pressing enter does not appear to cause a new file to load. As such I'm not sure why the text input even needs to exist. If it's just a lazy status indicator of current file loaded standard convention would be Titlebar. mysweetproject.nc - UGS. Note: In the Nightly I'm using, NetBeans appears to be in charge of the Title bar and app icon so apologies if the titlebar is accurate when this code reaches stable.

- [ ] Disable or remove the filename text box.
- [ ] Add filename to status bar.
- [ ] Add filename to title bar?

jahnj0584, 2017-09-04:
I'd keep it just as browse button and remove any function of the text
field. It was an annoyance the few times I accidently clicked upon it.

Put the file name in the visualizer along the bottom or something? Then you
can quickly see all relevant tool path information.

breiler, 2018-02-13:
I've tried to implement this as the following:

A new icon for the open action has been added to the toolbar and the old file browser is removed. If no file is opened it's now shown in the status bar:
[image]

An menu option has been added for the new action:
[image]

When a file is opened, the file name is shown in the title bar and on the status line:
[image]

If the workflow window is open, files are added to it. 
If the editor is open it will be reloaded with the new file.

jahnj0584, 2018-02-13:
neat

winder, 2018-02-16:
Resolved with #896

closed 2018-02-16 by winder