winder/Universal-G-Code-Sender #737: Usability Improvements: Visualizer

winder, 2017-09-04:
The top left cube looks like something from CAD software, but defies convention by being apparently non-interactive. Click face to aim camera directly at that plane, click and drag to move (which you don't need since your entire module has comparable functionality)

- [ ] Clickable corner-cube (with visual indicator of the highlighted face?).
- [ ] Visualizer toolbar actions with preset view angles.
- [ ] Double click to reset view angle to default.
- [ ] Remove `Jog To` button from right-click menu (or move it to a sub menu).
- [ ] Move right-click enable/disable options to a sub menu.
- [ ] Add the visualizer preset view angle actions (from above) to the right-click menu.

jahnj0584, 2017-09-04:
I'd opt for a view cube like how Solid works does it or the cube may be
enough as a datum instead. You only have to see the top, front, and left of
a tool path see what it's doing.

I'd keep the jog to control as right click. I find it very handy and a sub
menu makes it lose the convenience.

I would add a "get coord" command to see where a tool path physically will
go (XYZ value)

<https://github.com/winder/Universal-G-Code-Sender/issues/737>, or mute the
thread

breiler, 2018-02-18:
Added submenu for hiding/showing features in the visualiser sub menu:
[image]

breiler, 2018-02-18:
Added a submenu with view presets: Top, Left, Front and Reset (that will rotate to a isometric-like view):
[image]

jahnj0584, 2018-02-18:
Can you add those presets to the faces of the cube in the corner?

breiler, 2018-02-18:
It's on the todo-list, I'm picking the low hanging fruits right now. =P

breiler, 2019-04-26:
Note to self, I've been reading up on Ardor3D which could be a suitable scene graph library to use for implementing this other stuff: https://github.com/Renanse/Ardor3D