winder/Universal-G-Code-Sender #738: More advanced probing plugin

winder, 2017-09-04:
Most of the features in this video should be straight forward to implement: https://youtu.be/pQGnV85Cu_A?t=191

I'll probably go ahead and make a bunch of screens that go over the visualizer for these:
- [ ] Calibration (a.k.a. probe / end mill size measurement)
- [ ] Corner probing (inside / outside)
- [ ] Angle measuring.
- [ ] Single surface probing (X / Y / Z)
- [ ] Selectable work coordinate system for X & Y probing (G43.1 doesn't work or make sense for these).

vMeph, 2017-09-04:
That looks so good and promissing!!!!

jahnj0584, 2017-09-04:
Wicked. I guess we're heading towards a tabbed controller due to all the
new features? Could you make it in "workspaces" so we can have eg. DRO,
visualizer, file control, overrides in one then another workspace for
macros, probing, settings, etc? All user controlled of course ;)

aforww, 2017-09-05:
Fantastic

winder, 2017-09-07:
I updated the existing Z probe button tonight. Since the X/Y probing didn't work I replaced that setting with a retract field. There's also a surprise between the initial probe and the final retract that I won't spoil. :)

[image]

groberts22, 2017-09-09:
Here are a couple of thoughts on the probe function.

The current version of the probe function does not display the current units, which can be very confusing.
The two most common uses of the probe (at least for me) are
1. Resetting the z offset after a tool change, and 
2. setting the xyz zero values for a new workpiece.
For both of these, I use a touchplate with known xyz thickness values that sits directly on the wasteboard of the machine or fits over the corner of the workpiece.  I think this is what many/most people do when they get sick of using a piece of thin paper and a calculator.
What would be most helpful (to me) is to have a helper that would allow me to position the tool at a reasonable distance from the surface(s) of the touchplate and then have the machine automatically set the relevant zero values for me.
The display could look similar to the current one, with an option to set a single axis value only (typically z) or all axes at once, and an option to choose the target WCS offset (54-59). [It might also be necessary to allow separate thickness and distance/direction settings for x,y, and z.]
The function could then use the G38.2 and G10L20 commands.
Thanks for your consideration.

jahnj0584, 2017-09-09:
Keep in mind to do the x/y properly you cannot use an endmill. You'd have
to get a piece of known diameter turned steel rod and then swap again for
your Z

groberts22, 2017-09-09:
True, although I have found in practice that if I lower the probe as far as possible below the horizontal surface of the touch plate and make sure the bit is properly rotated I get reasonable accuracy and repeatability.

aforww, 2017-09-10:
Why would you have to do that? You absolutely can use the end mill. People
do it all the time

aforww, 2017-09-10:
Forgot to add, I was using it before X/Y was removed from the probing in the nighties.

jahnj0584, 2017-09-10:
Depends what accuracy you need, but you'd have to rotate the flute for each
axis you probe. Good luck with 0 flutes and all that. A 4 flute would be
easy enough to do though within reason. Still, nothing will repeat better
then a solid cylinder.

aforww, 2017-09-10:
Ah. Yea I can agree with that. I did it with a single O flute but had to
rotate it to catch the leading outside edge. Still faster than trying to
get it dead on by eye and a feeler gauge lol.

jahnj0584, 2017-09-12:
Can you get the macros and UI from this fellow open source sender? I'm trying it out tonight while we figure out the platform issues 

https://www.sourcerabbit.com/GCode-Sender/

winder, 2017-09-27:
I've been spending a lot of time working on re-designing the advanced probe page. My first attempt pretty much copied the [design in Mach4](https://youtu.be/pQGnV85Cu_A?t=264), but I wasn't happy with how it fit in UGP. Back at the drawing board I decided to make a fully 3D representation of the settings, and I'm really happy with how it came out. I can add additional tabs for some of the remaining advanced probe tabs.

[image]

[image]

[image]

vMeph, 2017-09-27:
@winder build #8​42 failed

Radiu, 2017-09-27:
looks very good, you think would be possible to come up with better visual layout design, similar to match or source rabit?

vMeph, 2017-09-27:
@winder  i dont see probing window now, only auto level?

winder, 2017-09-27:
If you used the auto update feature you might need to select a new module.
I moved the probe features to a separate one.

winder, 2017-09-27:
@Radiu could you make some specific suggestions? For this iteration I decided against modal dialogs like rabbit and static dialogs like Mach4, opting for a more dynamic representation in the visualizer.

vMeph, 2017-09-27:
@winder i have the autoupdate feature installed, just see there the auto level other probes i dont see,have looked arround and nothing

winder, 2017-09-27:
@vMeph The probe module might not be installed, is it listed under "Available Plugins"?

[image]

vMeph, 2017-09-27:
after last update cant connect to server grrr
[image]

winder, 2017-09-27:
Sourceforge seems to be having some trouble the last couple days https://sourceforge.net/projects/ugs-update-center/files/

vMeph, 2017-09-27:
ya thats what i tought

Radiu, 2017-09-27:
what i mean is more professional graphic design
something like this and more friendly use
[image]
[image]

apart from it is just a sugestion, your work is the best
thanks alot for all tthe work that you provide to us

vMeph, 2017-09-28:
that looks cool
im not good on designs but here is a quick design i made based on that just to get a idea
defenetly that the console space area, would be the perfect place to implemente interactive menus with some friendly and easy to folow structures and make UGS look a clean and modern design
@winder any ideas?

[image]

winder, 2017-10-25:
I made a video demonstrating the current Probe Module features:
[image])

vMeph, 2017-10-25:
very nice video

thanks for all the work

smallpoul, 2017-11-13:
Hi will,
           thanks for your very good program, i like and use a lot the probing module, the only problem that i have is when doing the x and y probing, for some reason, it does the x good but when it position the machine to seek the y, it never goes far enough to the right (trying to find the lower left zero) to be able to touch the plate. is there soming that i miss.
thanks
Mike

winder, 2017-11-14:
@smallpoul you can increase the "offset" values to make it probe further

smallpoul, 2017-11-15:
Sorry if I don't understand you right but the I was thinking that the offset was the tickness of you're border probe?

winder, 2017-11-15:
Oops. You're right. One of the configurable values should be the probe
distance though. You can set it to whatever you like. the visualizer will
show longer arrows when you have the right one.

smallpoul, 2017-11-15:
No problem, here the problem in code:
`>>> G91 G21 G0 X-14.7037
>>> G90 G21
>>> G91 G21 G0 Y-15.0
>>> G90 G21
ok
>>> G91 G21 G0 X15.0`

after probing x, it will go to the left for (x) -14.7037 then go back for (y) -15 and then go to the right for (x) 15. the problem is that when it place the end mill to seek the y, It goes only around .3 mm from the edge and the endmill does not go far enough to have a good contact with the block.
thanks again

winder, 2017-11-15:
Could you take a screenshot of the settings you're using and the entire console output of the routine (including the error).

smallpoul, 2017-11-15:
Here a video of the problem and the setting

https://www.youtube.com/watch?v=dDDf4rT1qO0

here s  the code

`
>>> G21 G91 G49
>>> G38.2 Z-15 F150
>>> G90 G21
ok
[PRB:0.3883,0.1396,0.0247:1]
ok
ok
>>> G91 G21 G0 Z1.0
>>> G90 G21
ok
ok
>>> G21 G91 G49
>>> G38.2 Z-15 F100
>>> G90 G21
ok
[PRB:0.3883,0.1396,0.0126:1]
ok
ok
>>> G91 G21 G0 Z0.1124
>>> G90 G21
>>> G91 G21 G0 X-15.0
>>> G90 G21
ok
ok
>>> G91 G21 G0 Z-15.0
>>> G90 G21
ok
ok
ok
ok
>>> G21 G91 G49
>>> G38.2 X15 F150
>>> G90 G21
ok
[PRB:-0.0157,0.1396,-0.5772:1]
ok
ok
>>> G91 G21 G0 X-1.0
>>> G90 G21
ok
ok
>>> G21 G91 G49
>>> G38.2 X15 F100
>>> G90 G21
ok
[PRB:-0.0162,0.1396,-0.5772:1]
ok
ok
>>> G91 G21 G0 X-14.595500000000001
>>> G90 G21
>>> G91 G21 G0 Y-15.0
>>> G90 G21
>>> G91 G21 G0 X15.0
>>> G90 G21
ok
ok
ok
ok
ok
ok
>>> G21 G91 G49
>>> G38.2 Y15 F150
>>> G90 G21
ok
[PRB:0.0029,-0.1116,-0.5772:1]
ok
ok
>>> G91 G21 G0 Y-1.0
>>> G90 G21
ok
ok
>>> G21 G91 G49
>>> G38.2 Y15 F100
>>> G90 G21
ok
[PRB:0.0029,-0.1121,-0.5772:1]
ok
ok
>>> G91 G21 G0 Y-14.7483
>>> G90 G21
>>> G91 G21 G0 Z15.0
>>> G90 G21
ok
>>> G91 G21 G0 Y15.0
>>> G90 G21
>>> G10 L20 P1 X-2.7955 Y-2.9483 Z0.1124
ok
ok
ok
ok
ok
ok
>>> G90 G21
ok
`

winder, 2017-11-15:
@smallpoul thanks, that looks like a bug. I'll try to fix this soon.

Jibeji, 2018-04-16:
Hi,
Is there any plan to have a Hole Center Finder module available?

timon2jkl, 2018-04-28:
When I XYZ edge find everything works fine and 0's everything but if I just do an Z it does not 0 the Z axis it sets it to the pull off distance plus edge finder and when you say return to 0 it plunges into the stock by the difference Am I doing something wrong or is it a Bug?  This is on a raspberry pi 2 G-shield April 22 nightly.

robllewellyn, 2018-04-29:
I am seeing similar output to @timon2jkl 

Get close, probe with 3mm brass plate on the Z. 

Sometimes retrun to zero will not go deep enough, sometimes it goes too deep into the stock...

Product Version: Universal Gcode Platform 201609300101

If you let me know what info is helpful, or if I am obviously not doing something right, let me know.

haaxema, 2018-05-01:
Hi all,

I am using UGS for a couple of weeks now and it works great.
I'd also like to see the Hole Center Finder module, is it in development or waiting for contributors?

winder, 2018-05-01:
@haaxema I don't think anyone is actively working on it. If you'd like to take a look, the code is here:

https://github.com/winder/Universal-G-Code-Sender/tree/master/ugs-platform/ProbeModule/src/main/java/com/willwinder/ugs/platform/probe

DavidArenillas, 2019-01-21:
the same for me. It seems that it doesn't keep the values...

CJsolar, 2019-04-25:
Hello, I'm new to the new version of USG Platform.  I down loaded (what I thought was the latest version) from GitHub yesterday, 4-24-19.  The Probe Module plug in is not in the software.  I hit the plug in update button and nothing new comes up, I checked for UGS updates and it says I have the latest version.  I then found and downloaded "Probe Module 2.0-Snaps.nbm" file and tried to install the plug in into UGS.  It said I couldn't install it because of some other file in UGS was an earlier version.  I dont know why this is so hard.  According to Will Winder in a You Tube video I watched on how to use the Probe Module, the USG Platform should have the plug in the software.  Is there a place where I can download USG Platform that has all the updates built into it?

Carl

tkms002, 2019-05-27:
I second this issue.  How do I get the PROBE MODULE??

kthornbloom, 2020-07-10:
@winder - Came across [this video](https://www.youtube.com/watch?v=bhZdbgM6S70&feature=youtu.be&t=60) recently which demonstrates some angle finding. The X,Y,and Z get zeroed as usual, and then it does an extra step to determine the angle of the material. No more attempting (and failing) to line it up perfectly with the machine. This would be super handy!

PixelTamer, 2020-08-07:
Hi, 
I make and sell an edge-finder that has a hole in it over the X0Y0 corner. I wrote a routine to find the center of the hole and set Z, X & Y to zero in MACH3.  The advantage is that you don't need to know the diameter of the probe. My UGS customers ask me how to create a similar macro to do that routine in UGS. Unfortunately, One needs to temporarily read and store a few variables to accomplish the routine, and UGS and GRBL just don't seem to offer that capability. 
What is the probability that a center finding routine can be added to the Probing module?

Thanks, 
Pixel
[image]

Threeshedz, 2021-10-07:
> Hi, I make and sell an edge-finder that has a hole in it over the X0Y0 corner. I wrote a routine to find the center of the hole and set Z, X & Y to zero in MACH3. The advantage is that you don't [...]

I would love to see a center finding routing in USG! I'd assume this could be achieved with a plugin?

https://winder.github.io/ugs_website/dev/plugin/