winder/Universal-G-Code-Sender #739: Status Plugin

winder, 2017-09-04:
The DRO plugin is doing double duty right now by showing active endstops, machine state and the latest comment. It also doesn't show all status information.

bCNC and GrblPanel both have nice little `state` displays for the following information:
Coordinate system (G54- G59)
Distance Mode (Absolute/G90, Relative/G91)
Units (inch/G20, mm/G21)
Feed Rate Units (this might be a grbl-panel setting)
Arc plane (XY/G17, ZX/G18, YZ/G19)
Current Feed Speed (Grbl 1.1)
Enabled endstops (Grbl 1.1)
Feed override (Grbl 1.1)
Spindle/Mist/Flood state

Each entry should include a combo box which allows changing the value.

jahnj0584, 2017-09-04:
Good call. In my mind, it should be the 3 axis as it is currently, zero
buttons on the right of the value, editable fields for the job coords (if
you're off by 1" you can change it to 21.7" instead of 20.7" and not have
to physically move the machine there)

States should just be a little light up text around the DRO values (look at
lathe DROs for example. They are all the same because the system works
well!)

The machine door lock and all those other commands need to be split since
they are not on many or all machines and waste screen real estate as it
stands.

groberts22, 2017-09-09:
I don't know about others, but I think it would be a HUGE improvement to have the items identified by @winder (particularly the first three) reported (and potentially editable) in the status (DRO) plugin. If it were an actual button, I would have worn out the "Get State" option on the common commands tab by now.
Also, I am not sure what the use case is for the items currently included at the bottom of the panel, or for the latest comment.
Finally, I think it would make a lot of sense to have the common commands and DRO plugins combined, as they are central to the value of the program (at least to me).
BTW, I think UGS is great - I only wish I had the know how to help in some way other than just commenting.  Thanks.

winder, 2017-11-16:
@jahnj0584 moved some things around here. Things are organized a little better now with real time things in the DRO window and state in a new State window.

[image]

closed 2017-11-16 by winder

carneeki, 2017-11-16:
@winder the preview image looks great! I just pulled the source and recompiled, but can't find the state panel anywhere. Is an additional commit on its way, or am I missing something?

Thanks!

winder, 2017-11-16:
@carneeki since this is a "multi module" maven project, you sometimes need to build with dependencies to pick up all changes (I think adding `-am` to the CLI does this). Alternatively `mvn clean` before your build to make sure everything is picked up.

carneeki, 2017-11-16:
@winder I tried adding the `-am` arg but couldn't get the project to compile (errors with org.codehause.mojo brandingToken - I'll investigate that tomorrow). Compiling as usual with `mvn clean` worked, but still no state panel. I also cleaned out `~/.ugs` just to be sure.

winder, 2017-11-16:
@carneeki Netbeans has its own cache directory that often causes problems, you can try clearing out as well. I'm a little surprised/disappointed that a brand new window is causing these problems. The NB cache usually only gets stale when renaming existing windows. When things are working there should be a `State` option in the `Windows` menu.

vMeph, 2017-11-16:
@winder  i did the update the controller state dro new tabs shows but i dont have no State new panel on windows
on the updates bottom i see there a problem says:
A java.util.MissingResourceException exception has occurred.
However, the system should continue working without further problems.
Click Show Details for the stack trace.

detail:

Annotation: Offending classloader: ModuleCL@27910354[com.willwinder.DowelModule]
Annotation: #149833: failed to find com.willwinder.ugs.platform.probe.Bundle in locale en_US in ModuleCL@27910354[com.willwinder.DowelModule] for C:\Users\vMeph\Desktop\ugsplatform\ugsplatform\modules\com-willwinder-DowelModule.jar; resource lookup of com/willwinder/ugs/platform/probe/Bundle.properties -> null
Annotation: Offending classloader: ModuleCL@27910354[com.willwinder.DowelModule]
Annotation: #149833: failed to find com.willwinder.ugs.platform.probe.Bundle in locale en_US in ModuleCL@27910354[com.willwinder.DowelModule] for C:\Users\vMeph\Desktop\ugsplatform\ugsplatform\modules\com-willwinder-DowelModule.jar; resource lookup of com/willwinder/ugs/platform/probe/Bundle.properties -> null
Offending classloader: ModuleCL@27910354[com.willwinder.DowelModule]
#149833: failed to find com.willwinder.ugs.platform.probe.Bundle in locale en_US in ModuleCL@27910354[com.willwinder.DowelModule] for C:\Users\vMeph\Desktop\ugsplatform\ugsplatform\modules\com-willwinder-DowelModule.jar; resource lookup of com/willwinder/ugs/platform/probe/Bundle.properties -> null
Offending classloader: ModuleCL@27910354[com.willwinder.DowelModule]
#149833: failed to find com.willwinder.ugs.platform.probe.Bundle in locale en_US in ModuleCL@27910354[com.willwinder.DowelModule] for C:\Users\vMeph\Desktop\ugsplatform\ugsplatform\modules\com-willwinder-DowelModule.jar; resource lookup of com/willwinder/ugs/platform/probe/Bundle.properties -> null
Caused: java.util.MissingResourceException: No such bundle com.willwinder.ugs.platform.probe.Bundle
	at org.openide.util.NbBundle.getBundle(NbBundle.java:451)
[catch] at org.netbeans.StandardModule.getLocalizedAttribute(StandardModule.java:169)
	at org.netbeans.modules.autoupdate.services.ModuleUpdateElementImpl.getRawCategory(ModuleUpdateElementImpl.java:182)
	at org.netbeans.modules.autoupdate.services.ModuleUpdateElementImpl.getCategory(ModuleUpdateElementImpl.java:194)
	at org.netbeans.api.autoupdate.UpdateElement.getCategory(UpdateElement.java:198)
	at org.netbeans.modules.autoupdate.ui.Utilities.makeInstalledCategories(Utilities.java:109)
	at org.netbeans.modules.autoupdate.ui.InstalledTableModel.setUnits(InstalledTableModel.java:87)
	at org.netbeans.modules.autoupdate.ui.InstalledTableModel.setUnits(InstalledTableModel.java:92)
	at org.netbeans.modules.autoupdate.ui.PluginManagerUI.refreshUnitsInBackground(PluginManagerUI.java:608)
	at org.netbeans.modules.autoupdate.ui.PluginManagerUI.initialize(PluginManagerUI.java:251)
	at org.netbeans.modules.autoupdate.ui.PluginManagerUI.access$100(PluginManagerUI.java:79)
	at org.netbeans.modules.autoupdate.ui.PluginManagerUI$1.run(PluginManagerUI.java:134)
	at org.openide.util.RequestProcessor$Task.run(RequestProcessor.java:1443)
	at org.netbeans.modules.openide.util.GlobalLookup.execute(GlobalLookup.java:68)
	at org.openide.util.lookup.Lookups.executeWith(Lookups.java:303)
	at org.openide.util.RequestProcessor$Processor.run(RequestProcessor.java:2058)

carneeki, 2017-11-17:
@vMeph I can confirm getting the same error you are.
@winder I just tried compiling on a computer that hasn't run UGS before, and still no State window (not even the menu item), so I'm not sure it's a cache problem.

winder, 2017-11-17:
@carneeki aah you're totally right. I committed the files which changed but forgot to add the new one! The other error is a bug, but unrelated. Both problems should be resolved now.

carneeki, 2017-11-17:
It's looking great!

Going to give it a test whirl now  :+1:

commit ba45a6f3 (2017-11-16): New state panel. Closes #739