winder/Universal-G-Code-Sender #740: Jog Control Overhaul

jahnj0584, 2017-09-04:
The whole step size and feed could be improved. I recommend a mix of what we have with what Haas machines do. 

Keys for step size. Its 0.0001, 0.001, .01 and .1 inches for imperial. Metric could be 0.01, .1 ,1, and 10mm
I also think units shouldnt be mixed up so easily, considering if you are using metric, you use it for everything. If you use imperial, its all inches. The toggle button just wastes space.  

On the Hass, at .1 inch, holding the button jogs at half the machines rapids speed. I suggest we use the machines rapid rate for the largest step size (because most CNCs are larger then a mill and need the speed) and then go down 50% per size (100%,50%,25%,10%) . The jog button can then also be clicked, which moves one step at the size selected, at full speed.

Please keep the Z jog button off the the side though and vertical as is. The haas put them all in a silly star shape which just doesnt make sense anymore. 

Would that be possible?

groberts22, 2017-09-09:
The jogging support functions on both bCNC and GrblPanel appear to map the keys on a standard numpad by default to the various jogging functions in a way that makes a lot of sense to me, and more or less implements what I described in my previous comment regarding the functionality of the UGS keymap option.

For example, in bCNC, the default keymappings are 

<Right> pan +X
<Left> pan -X
<Up> pan +Y
<Down> pan -Y
<PgUp> pan +Z
<PgDn> pan -Z
<+> or <=> Increase step by one unit (=current decade)
<-> or <_> Decrease step by one unit (=current decade)
<1> Set as step the step1 (default 0.1) value from the configuration
<2> Set as step the step2 (default 1) value from the configuration
<3> Set as step the step3 (default 10) value from the configuration
<*> Multiply step by x10
</> Divide step by /10

Jog mode is initiated by hitting the esc key, which ensures you do not inadvertently jog while trying to accomplish something else.

To me, being able to use the jog commands while standing at the machine using a wireless keypad (rather than sitting at the computer, which in my case is not particularly proximate to the machine) is critical.  What I am most interested in having on the display is a readout that shows the current values of the step size, feed rate, and unit variables.

winder, 2017-10-13:
superseded by #779

closed 2017-10-13 by winder