winder/Universal-G-Code-Sender #741: USB C issue on macbook pro

MoebiusFactory, 2017-09-05:
Hi,

When using UGS 2.0 on macbook pro with only the USB C to USB 2 adapter (the official one from apple as well as the dock from HooToo), i get this error message each time i try to connect.

On UGS 1.0.9 > i have a similar issue only if i have several usb devices connected to the USB C port (when i use the HooToo dock : 3 USB 2 port on 1 USB C). If i use the official apple adapter (1USB C to 1 USB 2 connector), UGS "sees" the right port and connects me without problem.

[image]

Thx a lot for making CNC milling possible trough your excellent software !

MF

winder, 2017-09-09:
When you see this problem try running `lsof | grep tty.usbmodem1411`, that should tell you what program is using the serial port. There are some more details about doing this here: https://www.baldengineer.com/arduino-fixing-serial-port-in-use.html

MoebiusFactory, 2017-09-12:
Hi winder !

Thx for the tip. I didn't get anything from terminal (weird i know !)
[image]

But, buy closing 3D printing softwares (that tries to capture the USB port all the time!), i solved my problem.

Thx again !

closed 2017-09-12 by MoebiusFactory