winder/Universal-G-Code-Sender #742: Z probing throwing error upon completion.

aforww, 2017-09-05:
Z probing working well now and applying offsets as expected. However, now we are throwing this error upon completion: Error Processing response and PRB:-192.200,-202.000,-40.400,1]>:-19.

RoadkillUK, 2017-09-10:
I get 2 errors, basically the same, I assume it's when the probe touches.  I used to use UGS(not platform) and I always had to 'Unlock' the machine after probing (using macros).

This is one such message. "Error while processing response [PRB:-488.000,-498.000]>:-48"

winder, 2017-09-11:
@RoadkillUK if you could run UGS from the console there should be a detailed error message about where the problem originated. I haven't been able to reproduce the problem at this point, so that message would be very useful.

Also, did you copy/paste that error message or type it in? The probe should report X/Y/Z but there is only 2 sections there.

RoadkillUK, 2017-09-11:
Hey @winder , thanks for the reply, it's 1:10 am Monday morning here, I should be in bed already, just about to go. I'll give it a go tomorrow after work.  As for the error message, I copied and pasted it ... I think.

We'll have another look tomorrow :)

jahnj0584, 2017-09-11:
I had this probe error this morning. It a working now when I wanted to get the code for you. Try clearing your eeprom (first copy your $$ to notepad) then delete all arduino libraries and flash GRBL 1.1 again. It worked for me

RoadkillUK, 2017-09-11:
Here is the console log from switchon.  I don't know if it makes any difference but I'm using WinXP!

**** Connected to COM4 @ 115200 baud ****

Grbl 1.1f ['$' for help]
>>> $$
>>> $G
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 11    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 11    (Homing direction invert, mask)
$24 = 50.000    (Homing locate feed rate, mm/min)
$25 = 1500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 2.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 800.000    (X-axis travel resolution, step/mm)
$101 = 800.000    (Y-axis travel resolution, step/mm)
$102 = 2560.000    (Z-axis travel resolution, step/mm)
$110 = 2500.000    (X-axis maximum rate, mm/min)
$111 = 2500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 30.000    (X-axis acceleration, mm/sec^2)
$121 = 30.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 490.000    (X-axis maximum travel, millimeters)
$131 = 475.000    (Y-axis maximum travel, millimeters)
$132 = 60.000    (Z-axis maximum travel, millimeters)
ok
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
>>> $X
[MSG:Caution: Unlocked]
ok

** THIS IS WHERE I PRESS THE INITIATE PROBE BUTTON **

>>> G21 G91 G49
>>> G38.2 Z-20 F75
>>> G90 G21
ok

** PROBE TOUCHES, THEN THIS ERROR (SOMETIMES IT DOES THIS TWICE AS IT'S MEANT TO, BUT NOT TODAY ** 

Error while processing response <[PRB:0.000,0.000,-5.379:1]>ok
ok

** THIS IS WHERE I TRY AGAIN, INITIATE PROBE BUTTON **

>>> $J=G21G91Z10F500
ok

** NOTHING HAPPENS, SO I ISSUE THE FOLLOWING COMMAND AND THE PROBE GOES DOWN **

>>> G38.2 Z-30 F70

** PROBE TOUCHES, THEN THIS ERROR **

Error while processing response <[PRB:0.000,0.000,-5.476:1]>ok

** I ISSUE THE FOLLOWING COMMAND TO SET MANUALLY **

>>> G92 Z19.33
ok

** WHEN I ISSUE THIS COMMAND MANUALLY THE PROBE IS STILL TOUCHING **

>>> z10 F300
(ALARM:4) Probe fail. Probe is not in the expected initial state before starting probe cycle when G38.2 and G38.3 is not triggered and G38.4 and G38.5 is triggered.(ALARM:4) Probe fail. Probe is not in the expected initial state before starting probe cycle when G38.2 and G38.3 is not triggered and G38.4 and G38.5 is triggered.
Error while processing response <ok>: An unexpected command was completed by the controller.

RoadkillUK, 2017-09-12:
@winder Here, have another log, this one is verbose (you don't say).

From connection to error.

**** Connected to COM4 @ 115200 baud ****

Grbl 1.1f ['$' for help]
>>> $$
>>> $G
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 11    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 11    (Homing direction invert, mask)
$24 = 50.000    (Homing locate feed rate, mm/min)
$25 = 1500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 2.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 800.000    (X-axis travel resolution, step/mm)
$101 = 800.000    (Y-axis travel resolution, step/mm)
$102 = 2560.000    (Z-axis travel resolution, step/mm)
$110 = 2500.000    (X-axis maximum rate, mm/min)
$111 = 2500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 30.000    (X-axis acceleration, mm/sec^2)
$121 = 30.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 490.000    (X-axis maximum travel, millimeters)
$131 = 475.000    (Y-axis maximum travel, millimeters)
$132 = 60.000    (Z-axis maximum travel, millimeters)
ok
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
>>> $X
[MSG:Caution: Unlocked]
ok
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z|WCO:0.000,0.000,0.000>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z|Ov:100,100,100>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z|WCO:0.000,0.000,0.000>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z|Ov:100,100,100>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Pn:Z>
[... 76 lines cut ...]
[verbose] <Run|MPos:0.000,0.000,-12.303|Bf:14,120|FS:75,0>
[verbose] <Run|MPos:0.000,0.000,-12.552|Bf:14,120|FS:75,0>
[verbose] <Run|MPos:0.000,0.000,-12.802|Bf:14,120|FS:75,0>
[verbose] <Run|MPos:0.000,0.000,-13.051|Bf:14,120|FS:75,0>
[verbose] <Run|MPos:0.000,0.000,-13.301|Bf:14,120|FS:45,0|Pn:P>
Error while processing response <[PRB:0.000,0.000,-13.243:1]>ok
ok
[verbose] <Idle|MPos:0.000,0.000,-13.375|Bf:15,128|FS:0,0|Pn:P>
[verbose] <Idle|MPos:0.000,0.000,-13.375|Bf:15,128|FS:0,0|Pn:P>
[verbose] <Idle|MPos:0.000,0.000,-13.375|Bf:15,128|FS:0,0|Pn:P>
[verbose] <Idle|MPos:0.000,0.000,-13.375|Bf:15,128|FS:0,0|Pn:P>
[verbose] <Idle|MPos:0.000,0.000,-13.375|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:0.000,0.000,-13.375|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:0.000,0.000,-13.375|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:0.000,0.000,-13.375|Bf:15,128|FS:0,0|WCO:0.000,0.000,0.000>
[verbose] <Idle|MPos:0.000,0.000,-13.375|Bf:15,128|FS:0,0|Ov:100,100,100>
[verbose] <Idle|MPos:0.000,0.000,-13.375|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:0.000,0.000,-13.375|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:0.000,0.000,-13.375|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:0.000,0.000,-13.375|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:0.000,0.000,-13.375|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:0.000,0.000,-13.375|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:0.000,0.000,-13.375|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:0.000,0.000,-13.375|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:0.000,0.000,-13.375|Bf:15,128|FS:0,0|WCO:0.000,0.000,0.000>

winder, 2017-09-12:
What I really need is the OS console log, are you on windows? If you run
the .bat file from the CMD application there will be a much more detailed
error message.

RoadkillUK, 2017-09-12:
@winder I have no idea what you mean, what .bat file, which application. Also WindowsXP. Cheers.

RoadkillUK, 2017-10-05:
I think this is a problem with the hardware (grbl/arduino) as it worked fine after a reinstall but happened again after I reinstated my settings.

winder, 2017-10-13:
@RoadkillUK could you try using the platform build? When the error happens the detailed error message occurs there is an icon at the bottom with the detailed error message I need to investigate this problem. See the screenshot below.

[image]

aforww, 2017-10-13:
Seems this is the same error I had sent you an email on.no?

winder, 2017-10-13:
@aforww the ones you sent me were related the SurfaceScanner, which has so many other problems it might have obscured this more specific issue around parsing the probe result. If you're still seeing this issue with singular probes any data you can get me would help. If you can run the `G38.2` command manually and reproduce the problem, that would be even better.

aforww, 2017-10-13:
Mine was both issues lol. But unfortunately can't be of much help right
now. Moved up to a larger machine and completely redid all my electronics
to accommodate a VFD spindle and Gcode spindle control using GRBL. Still
have some wiring and VFD programming to deal with.

RoadkillUK, 2017-10-13:
@winder Here you go, I just typed in the console G38.2 F50 Z-20 and here is the result..

[Imgur files.](https://imgur.com/a/YHDVB)

winder, 2017-10-14:
@RoadkillUK @aforww This should be resolved now, let me know if you continue to see issues with basic probe operations.

closed 2017-10-14 by winder

RoadkillUK, 2017-10-15:
Problem resolved, thanks.  It was nice to be able to Zero without cancelling 2 errors. :)