winder/Universal-G-Code-Sender #743: Error while starting file stream : null

scottcpledger, 2017-09-06:
Hello, I am hitting a problem. 
here is the run down. 
i loaded a file, and this file loaded fine the first time. 
i went in again and tried to load it , this time , nothing pulled up on the visualizer . I thought this was a little weird but tried to hit the play button anyways and try to go ahead and make my carve. 
when i hit play a little box popped up stating " Error while starting file stream:null" . 
i thought maybe the file got corrupt , so i tried to recreate it , 2 different times with the same result. 
I have seen issue#214 but that is a bit old and my version of software is newer than what was mentioned there. I am running version 2.0 nightly / jul 23,2017

thanks

scottcpledger, 2017-09-06:
oh , and it appears other files are fine , it is just this one file, that just went wonky after the second time trying to open it.

winder, 2017-09-06:
Could you share the file? You might have found an edge case that causes a problem in the gcode parser

closed 2017-10-13 by winder

scottcpledger, 2017-10-13:
Will.  Prey for not getting back to u.  I am out of town for the week but I
will get it to u next week.  I found a workaround.  Basically when I would
set the plunge rate along with multiple passes I get the problem.  If I
leave the plunge rate at 0 I do not see the issue.  Again it's only when I
do multiple passes.  I'll have the file to u next week.  Thanks