winder/Universal-G-Code-Sender #744: Cannot find TopComponent with preferedID AutoLevelerTopComponent

oddish2211, 2017-09-06:
Morning,

I tried using the autoleveler feature but unfortunately it immediately presents me this stack trace.
This is build c1e66de67ce077d16fb7294e3c9d340d2ff670cd

`java.lang.IllegalStateException: Cannot find TopComponent with preferredID AutoLevelerTopComponent, see IDE log for more details.
	at org.openide.windows.OpenComponentAction.actionPerformed(OpenComponentAction.java:92)
	at org.openide.awt.AlwaysEnabledAction$1.run(AlwaysEnabledAction.java:199)
	at org.openide.util.actions.ActionInvoker$1.run(ActionInvoker.java:95)
	at org.openide.util.actions.ActionInvoker.doPerformAction(ActionInvoker.java:116)
	at org.openide.util.actions.ActionInvoker.invokeAction(ActionInvoker.java:99)
	at org.openide.awt.AlwaysEnabledAction.actionPerformed(AlwaysEnabledAction.java:202)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:877)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at org.netbeans.core.TimableEventQueue.dispatchEvent(TimableEventQueue.java:159)
[catch] at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
`

oddish2211, 2017-09-06:
Oh apparently this is the result of a ClassCastException on line 142 of AutoLevelerTopComponent.java
als.zSurface gets boxed to a Double and then casted to an Integer internally in Swing.

oddish2211, 2017-09-07:
Solved with commit #745

closed 2017-09-07 by oddish2211

ajmendez, 2017-10-07:
I am not sure if something was reverted, but I started seeing this error on the Mac OSX nightly build from Oct 04, 2017. I will try a more stable version to see if that fixes things.

java.lang.IllegalStateException: Cannot find TopComponent with preferredID VisualizerTopComponent, see IDE log for more details.
	at org.openide.windows.OpenComponentAction.actionPerformed(OpenComponentAction.java:92)
	at org.openide.awt.AlwaysEnabledAction$1.run(AlwaysEnabledAction.java:199)
	at org.openide.util.actions.ActionInvoker$1.run(ActionInvoker.java:95)
	at org.openide.util.actions.ActionInvoker.doPerformAction(ActionInvoker.java:116)
	at org.openide.util.actions.ActionInvoker.invokeAction(ActionInvoker.java:99)
	at org.openide.awt.AlwaysEnabledAction.actionPerformed(AlwaysEnabledAction.java:202)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at com.apple.laf.ScreenMenuItem.actionPerformed(ScreenMenuItem.java:125)
	at java.awt.MenuItem.processActionEvent(MenuItem.java:669)
	at java.awt.MenuItem.processEvent(MenuItem.java:628)
	at java.awt.MenuComponent.dispatchEventImpl(MenuComponent.java:351)
	at java.awt.MenuComponent.dispatchEvent(MenuComponent.java:339)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:749)
	at java.awt.EventQueue.access$400(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:697)
	at java.awt.EventQueue$3.run(EventQueue.java:691)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:719)
	at java.awt.EventQueue$4.run(EventQueue.java:717)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:716)
	at org.netbeans.core.TimableEventQueue.dispatchEvent(TimableEventQueue.java:159)
[catch] at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

winder, 2017-10-08:
@ajmendez if re-downloading doesn't fix the problem you can clear the cache directory to fix the problem. The about page lists the location of that directory, on my machine it looks like this:
`Cache directory: /Users/wwinder/Library/Application Support/ugsplatform/2.0-SNAPSHOT/dev/var/cache`