winder/Universal-G-Code-Sender #747: Raspberry Pi 3 CPU Usage

BusbyGT, 2017-09-06:
I've noticed when using UGS on the Raspberry Pi 3 that the CPU utilization seems to cap out at 25%.

The Pi 3 is a quad core. It appears that UGS only ever uses 1 core of the CPU. When loading an 11MB Gcode file (600K+ lines), it pegs the CPU out on 1 core and only 1 core and it takes a few minutes to load the file (which is semi-understandable) but I kept help but wonder if the single core usage is the source or at least is a hindrance. Even during carving, it's maxing out a single core of the Rpi which just feels like its being restricted and could operate better.

I've seen this behavior on both Classic and Platform.

I'm going to do a test on my Windows laptop with a really really big file and see if I see the same behavior. I tried with the same 11MB file but the increased processing power of the laptop didn't really make it flinch.

I've tried searching the web for tips and tricks for the Java run time settings but basically kept coming up with nothing.

I don't know if this is necessarily a bug, an actual issue, or just an artifact of running on the Raspberry Pi 3.

winder, 2017-09-06:
There are a lot of things about processing gcode and streaming data to GRBL that can't really be parallelized.

Gcode is a big state machine, so you can't process commands without first processing the commands before it. Since UGS preprocesses the gcode before streaming it this can take a while for large programs (especially on slower machines like the RPi).

Similarly, the GRBL responses need to be processed in order. There are parts about the UGS event system which could be improved though, a "command complete" message might be sent to 20 different GUI elements so processing them in a thread pool could be a nice performance boost. I'm not sure why that would be pegging the CPU though, streaming is supposed to be pretty lightweight compared to the initial processing.

BusbyGT, 2017-09-07:
@winder This might be a Raspberry Pi specific issue. Maybe JRE related?

I opened an 80MB file in Windows 10 running Platform 64-bit:

[image]

It's using all four cores during processing. I see CPU going between 50 and 75% and thats total of all 4 cores. It ended up generating an out of memory error.

breiler, 2020-10-02:
Can't reproduce problem, closing issue

closed 2020-10-02 by breiler