winder/Universal-G-Code-Sender #749: Failure with Testing firmware

jahnj0584, 2017-09-06:
[image]

winder, 2017-09-07:
What were you doing when this happened?
In the lower right of the GUI status bar, is there a red error marker?
Are all the error messages the same message?

jahnj0584, 2017-09-07:
I was trying to simply connect to get my grbl settings without an arduino
plugged in, so I assumed that was the firmware I had to "connect" with

jahnj0584, 2017-09-07:
Yes it's all the same error message

winder, 2017-09-07:
I'm not exactly sure why this happened, but I it happened to me just now. There's a rare race condition somehow is my guess. It's easy to check for though so I should have a fix shortly, let me know if you see this again.

commit 9da0bd75 (2017-09-07): Fix strange null pointer exception. Closes #749
closed 2017-09-07 by winder