winder/Universal-G-Code-Sender #751: File load error Pop-up Suppression

BusbyGT, 2017-09-07:
When opening this file with UGS Platform on Raspberry Pi, there is no error pop-up, the UI just sits idle. However, if I click the Connect/Disconnect button after it, the error (seen below from Classic) will continually popup on every click.

[Kodiak-129644.txt](https://github.com/winder/Universal-G-Code-Sender/files/1283321/Kodiak-129644.txt)

When loading this file on UGS Classic, this error pops up:

[image]

UGS Platform on Windows pops up an error when loading the file.

closed 2017-09-08 by winder

BusbyGT, 2017-09-20:
@winder This is not fixed as of the September 17, 2017 build.

Loading a file that throws a parsing error, no window pops up BUT if you click on "Connect" or "Disconnect", it'll pop up the error window.

winder, 2017-09-20:
@BusbyGT thanks for the update. There are going to be a lot of improvements to this very soon which should make the parser (and error messages) much more accurate.

commit e93ffd97 (2017-09-08): Better gcode verification. Closes #751