winder/Universal-G-Code-Sender #752: Nightly Build Platform vs. Classic GUI

Brahe1, 2017-09-08:
Hello,

Sorry I am newbie with this stuff but I saw that grbl 1.1f came out (previously running 0.8c) and I wanted to try out the newest UGS platform. When I use the most current UGS Classic GUI it works perfectly fine. But whenever I use the the Platform, which it connects to just fine, I cannot control the motors. Before I was getting no response when I would hit any of the axis buttons, but now I am getting a humming sound from them but no movement. 

I figured out the problem was from the command lines. 

In the Classic GUI all I  would have to do is type or press a button:
       "**G91 G0** (axis then distance)"
But in the Nightly Build, pressing a button it comes out:
       "**$J=G21G91X-1F10**" (I understand that G21 and F10 are telling it how fast and stuff like that)

I can also type in the command like the Classic GUI into the Platform version and it will work but I want it to work by pressing the buttons.

Basically my question is, how do I get my machine/ardiuno to run off the new Platform with the new commands?

P.S. I know it is silly to ask but I am really not sure how to fix it.

Thank you,
Patrick

winder, 2017-09-08:
@Brahe1 the $J command is the new syntax for grbl jog motions. It is possible your acceleration settings are too high which could cause steppers to humm instead of moving.

The classic and platform GUIs both use the same backend code, so the differences are mainly the UI. The classic GUI will stay around as a simple UI, while more advanced features are being developed for the platform.

Brahe1, 2017-09-08:
I will look into the steps and mm/min when I can, but I believe they might be set already. Are there any other known issues?

winder, 2017-09-09:
There are a lot of common issues people have while setting up and configuring grbl. Some of them are documented here: https://github.com/gnea/grbl/wiki/Known-Issues

closed 2017-09-09 by winder

Brahe1, 2017-09-10:
@winder thank you for the help. I figured it was my feedrate! It is working perfectly now! Really appreciate it.