winder/Universal-G-Code-Sender #755: UGS breaks lines when sending.

FKjolner, 2017-09-10:
A file with the only line
`G90 G53 G0 Z-3`
is sent like this
```
G90 
G53 
G0 Z-3
```
This makes the machine "skip" the G53 (machine space) , instead acting the G0 Z-3 in G54 (work space)

If this line breakage is intentional, it needs some tweaking, either in general or for my board (xpro v3 with grbl1.1)

Any ideas?

Regards Fredrik

FKjolner, 2017-09-17:
I found this in the settings, but no info on how to make the option available to me.
[image]

Any Ideas?

Regards Fredrik

winder, 2017-09-19:
This problem exists for a number of non-modal commands. Currently, that command splitter just looks for `G` which isn't right. Unfortunately, the command splitter is needed for other reasons (thus not being able to disable it). I'll need to look into this.
```
   - Non-Modal Commands: G4, G10L2, G10L20, G28, G30, G28.1, G30.1, G53, G92, G92.1
```

FKjolner, 2017-09-19:
Thanks, I've been able to disable the command splitter in the .ugs\firmware_config\grbl.json but not found time to test it. 

For what reasons are the  command splitter needed? it might not impact my code.

regards Fredrik

closed 2017-09-27 by winder

cdsmith66, 2018-03-29:
I'm not sure if this is the proper place to leave this comment, so I apologize in advance. 

I was running an older version of UGS on a Raspberry Pi 3 and came across this issue, so I downloaded your nightly build yesterday(3/28) and loaded that on to the Pi. However I'm still experiencing this issue and the command splitter setting is still appearing. 

I have confirmed that a fresh install on a windows system does not have this setting/issue. I still have the older UGS files on the Pi, so is it possible that the new nightly build is referencing an old file? Thanks for the help.

Regards, Chris

whitetd, 2018-03-29:
What are you using to create your gcode? Use a preprocessor to break apart
the commands, that way if you have an error it does not ignore everything
after it.  It can also delete bad commands, and convert macros to usable
code. Like drilling macros to individual drill operations.

cdsmith66, 2018-03-29:
@whitetd I'm using HSM Express to generate the gcode and I don't think the gcode is the issue because the same file runs perfectly fine using the windows version of UGS.

cdsmith66, 2018-03-29:
@winder, so, I realized that i was using the "somewhat stable" version from July 28th which was before this issue was solved, my bad. However, the reason I was using that version was because I couldn't get the newest nightly build to run, similar to issue #918.