winder/Universal-G-Code-Sender #756: Files no longer working with same Post

jahnj0584, 2017-09-10:
[image]

Same kinda stuff I always ran

vMeph, 2017-09-10:
did you check in tools\options\controller options\regular expressions patern remover
if by mustake you have put there certain expressions to be removed?

jahnj0584, 2017-09-10:
It's not that

vMeph, 2017-09-10:
on past i used the fusion360 post process from Strooom/GRBL-Post-Processor witch i never add problems with files until i made a design that produced alot of those error:33 i made a issue on grbl gnea area to try figuer out what was the problem, it turns out that in my case the problem was from Strooom/GRBL-Post-Processor

the discussion about its here
https://github.com/gnea/grbl/issues/230

winder, 2017-09-10:
Do you get the error in check mode as well? Could you share the file so I can try reproducing it?

I made some tweaks to the gcode parser with #751. It wasn't supposed to change how gcode was modified though, all the files I tested before and after were identical but it's possible I missed something

jahnj0584, 2017-09-10:
Yep it comes up. In checking "allow helical moves" made on file work but if
I reposted it again itd fail.

I also have a Gcode here from last year. If I repost it, the new one fails.
This is either a GRBL or UGS issue because my post, CAD, and CAM (heck, the
whole computer) hasn't been used since November.

winder, 2017-09-11:
@jahnj0584 please attach one of those files on GitHub or email it to me, I'm not seeing this problem with any of my test programs.

winder, 2017-09-11:
This might be the same issue as #727, so it might not be related to the parser changes I made this week.

jahnj0584, 2017-09-11:
Check your email

jahnj0584, 2017-09-11:
Will this is a huge issue for me right now. Ive been building this CNC since april and now I cant because UGS just has issues with all of my CAM. Any way I can download a nightly build from last week instead?

winder, 2017-09-11:
@jahnj0584 could you try the latest stable build? It's about 6 weeks old now: http://winder.github.io/ugs_website/download/

I tried your file on my machine and wasn't able to reproduce the problem, I'll try upgrading grbl on my test board when I have a chance but if you could verify that the problem still occurs on older builds that would help narrow the problem down.

jahnj0584, 2017-09-11:
> website/download/

I will download it and try it tonight when I get back. I feel like thats
the one I have already though (best way to check version?)

winder, 2017-09-11:
@jahnj0584 if you download a version it will be in the title. If you're using the auto-updates it will be in the plugin (all UGS plugins use the same version right now).

[image]
[image]

jahnj0584, 2017-09-12:
Sep10 doesn't work. The Semi-stable platform doesn't even launch after the netbeans thing comes up. Crashes after loading modules. 

August 28 is the sender Im using and had this error to begin with.

vMeph, 2017-09-12:
Sep10 release working ok here, maybe need to clean cache?
just guessing

jahnj0584, 2017-09-12:
Reinstalling Java and restarting the PC. Well see what happens

winder, 2017-09-12:
If downgrading the platform version isn't working please try one of the
classic builds. They have use same processing code.

I wasn't able to reproduce the problem with the file you sent me, so seeing
if an older version of UGS works is really important to figuring out if
this is a grbl or UGS regression.

jahnj0584, 2017-09-12:
Java is freshly installed. 

Aug28 platform sends 33 and 36 errors
Sep 11 2016 won't open 
Sep 22 and dec 17 2016 crashes after "done loading modules"
Sep 8 2016 nightly fails same way

winder, 2017-09-13:
Were you using check mode in your tests?
Do you get the same error if you run the program with your motors and spindle turned off?
What controller hardware are you using?
Were you testing with the same "2D Pocket1.nc" file you sent me?
Do you get the same error with other gcode senders?

From what you're saying this problem has existed for the past 2 years, so I'm wondering if something else is going on with your machine.

winder, 2017-09-13:
What version of GRBL are you running? Any customizations to it?

jahnj0584, 2017-09-13:
Initial errors came when I tried sending file regularily. Everything today has been in check mode. The SR Gcode sender I linked today ran every file without issue or error. 

No difference if I have my control box on or not. My spindle and strippers are all shielded. My USB cable is also insulated and shielded. My PC power is shielded. Noise was never an issue in my shop. 

Hardware? Just an arduino. Same one as back in November with the last cnc. 

Straight up GRBL 1.1 using the hex from the gnea github page. Should I try a new flash maybe? I doubt that's the issue though. 

Yea I used that file as the consistent benchmark tonight.

jahnj0584, 2017-09-13:
The other weird thing is I had a posted Gcode that UGS will send without issue. When I re-posted it with new feed rates, it also came up with errors. The PC hasn't been touched since November (when I had my other cnc and workshop) so nothing has changed with my CAD, CAM, or post settings. Only new change is GRBL 1.1 and the sender version

jahnj0584, 2017-09-13:
I did a total clean flash of GRBL 0.9 and it still has an error in aug 28 platform

winder, 2017-09-13:
@jahnj0584 you have me convinced that there is a problem in UGS, thanks for going through all the testing effort.

Could you email me your "users/documents/.ugs" directory? This only has internal UGS settings in it, I want to make sure there is nothing strange about your settings.

If I'm understanding correctly:
1. UGS 1.0.9 and a separate gcode sender work. (Not a problem with the file)
2. You see the same error with GRBL 1.1 and GRBL 0.9, when using UGS 2.0. (Not a grbl regression)
3. UGS Platform and UGS Classic are broken for all tested 2.0 builds going back until at least 1 year ago. (Broken for a long time)

The problem for me is that the file you sent me works on my machine. So I want to install your settings and see if that can help me reproduce the issues.

jahnj0584, 2017-09-13:
It's sent. I also included the Gcode that I'm always using as a benchmark
here so we're on the same page

winder, 2017-09-13:
@jahnj0584 thanks! We're finally in business, I was able to reproduce the problem using your settings!

winder, 2017-09-14:
@jahnj0584 the "Decimal Truncator" setting is too low in your settings. Change it from 3 to 4 (the current default) in the "Controller Options" section of the Options page and the check mode completes with no errors!

In hindsight, I should have seen that it was an arc with an error:33 in your first screenshot and noticed that there were only 3 decimals of precision.

jahnj0584, 2017-09-14:
Ugh whyyyy. Something so simple. Weird that you'd need beyond 2 decimal places for mm's though? Why did an old file work and a new one didnt?

jahnj0584, 2017-09-14:
Should you lock that option in new versions?

jahnj0584, 2017-09-14:
[image]

Decimals are at 4. Using GRBL 0.9 and it shoots that error. I can hit send again and the file resumes but I have no idea if it skipped a line of code or not to do so. It had no problem with the same helix that happens on the second Z pass.

winder, 2017-09-14:
Try disabling all the regex patterns besides the M6

jahnj0584, 2017-09-14:
That what already cleared out

winder, 2017-09-14:
You should get a popup when that error happens which tells you the line that had an error. With no "regex" processors it fails because of the M6:
[image]

jahnj0584, 2017-09-15:
Ok so another three hours in the shop tonight. 

Sometimes It shoots a blank error. I hit resume and the rest of the file sends fine. 

Later tonight I run a simple code to cut a straight line (testing depths and speeds) and on the 6th time, the 33 and 20 errors pop up again. I only changed a Y value from 3.963 to 4.5 (and then 4.500) but those errors still popped up.

jahnj0584, 2017-09-15:
[image]

That's the code. Very basic. Line 19 is the one that errors 33

winder, 2017-09-15:
If the error didn't start happening until you modified the Y value then
this is a real error. You can't change the X/Y values of an arc that way
because of the way the IJK values work:
http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g2-g3

Error 33 means that the IJK values are incorrect for the provided XYZ.

jahnj0584, 2017-09-15:
Oh ok. Didn't know that. I'll repost the file to what it should be and test
it after work.

Thanks

jahnj0584, 2017-09-16:
I haven't had this issue in a while. May I suggest you lock the decimal truncate feature from here on out? Does changing it to 5 help anyone?

commit d95becf8 (2017-09-28): Make sure the decimal truncation processor has at least 4 decimal places. Closes #756
closed 2017-09-28 by winder