winder/Universal-G-Code-Sender #757: Purgatory pause commands

jahnj0584, 2017-09-11:
Paused file during safety retract to check on bit. Issued an M5 and it didn't do anything so issumed it was thrown out by the GUI. Sadly, when I hit resume the M5 was sent through and broke my bit since the spindle stopped. I noticed it slowing down but didn't hit the stop button (it needs to be bigger ;) ) in time. 

Can you make it so commands are either executed or not at all? Maybe only certain commands could be allowed (like M5,M3,coolant toggle but void all movement ones for safety reasons) or have them completely ignored and have no effect. 

Cheers

winder, 2017-09-11:
I'm not sure what you mean, If you pause the machine the resume button should send the remaining commands.

Maybe the "Command" text box should be disabled to avoid misleading the user

jahnj0584, 2017-09-11:
I sent the M5 manually through the console. It wasn't in the code.

winder, 2017-09-11:
You can't send (or insert) commands while the machine is paused, so I
should disable the console.

jahnj0584, 2017-09-11:
I'm saying maybe we should be able to for M codes, since UGS does not have
buttons like a Haas for coolant and spindle toggle, which are often used
mid-program. The GUI could then insert the opposite toggle before resuming
the files Gcode.

If that's too complicated, just disable the console completely.

winder, 2017-09-11:
Grbl has spindle/coolant/mist toggles in v1.1, they are on the Overrides
page. You can send them while the program is running

jahnj0584, 2017-09-11:
> issues/757#issuecomment-328387419
> AQlzDEWngPcDXB1DQeNcLd1GhuiBJ1Axks5shIIzgaJpZM4PSgij

Yes but that's no good if you're trying to take off an aluminum chip that's
stuck to your endmill. I need the machine paused.

jahnj0584, 2017-09-16:
Oh I see what you mean now. That works for me. Disabling the console while a file is in the buffer is the way to go then.

commit 4005a0e6 (2017-09-28): Disable console when machine is not idle. Closes #757
closed 2017-09-28 by winder