winder/Universal-G-Code-Sender #758: Optional Nightly Build pop up message

jahnj0584, 2017-09-11:
Can you make this have an option such as "never show again" or something? I've opened UGS at least a dozen times today due to controller issues and leaving the work shop and it gets a bit annoying ;)

vMeph, 2017-09-11:
Yes i would love this to have a option to show or not show pop ups at start.
Its very annoying to have the pop ups, i know they there to warn if this or that is experimental but is a nightly build so i already know that can have some bugs etc..
A option to show warnings or not would be ideal

winder, 2017-09-11:
This actually exists, but I never put it in the UI.

On mac it is the file `~/Library/Preferences/ugs/UniversalGcodeSender.json`
on windows it should be `/home/user/.UniversalGcodeSender.json`
and on linux `~/.UniversalGcodeSender.json`

Change `"showNightlyWarning": true,` to `false`

jahnj0584, 2017-09-11:
For people with issues, notepad++ is able to open this file.

jahnj0584, 2017-09-11:
It's on line 35

vMeph, 2017-09-11:
Im not able to find the directory
Wouldnt then need to have a ide to be able to make those changes?

Wouldnt be possible to set this option under options somewhere with a box to check or something, it would be more easy for all users that dont want to see pop ups at start

vMeph, 2017-09-11:
Where is it on windows?

jahnj0584, 2017-09-11:
It's in users/documents/.ugs. You may have to enable hidden files

vMeph, 2017-09-11:
Ok found it

vMeph, 2017-09-11:
I still get the help pop up
This is a experimental feature.please use with caution
Report any bugs you find on github

jahnj0584, 2017-09-11:
I left the comma after false and saved the file. Maybe you forgot

vMeph, 2017-09-11:
usualy comes 2 pop ups one about nightly build and other about experimental feature
the nightly one disapear but not the experimental feature

jahnj0584, 2017-09-11:
I only ever had the nightly message...

winder, 2017-09-11:
Close the auto leveling window to get rid of the second one

commit d337a0f9 (2017-10-13): Expose 'showNightlyWarning' setting. Closes #758
closed 2017-10-13 by winder