winder/Universal-G-Code-Sender #759: No jog controller on restart

jahnj0584, 2017-09-11:
Set grbl to $c. Send a file that failed with error 3 or whatever it is were already trying to solve. 
Soft reset still won't let me hit send as it's "currently streaming" even though file is done. If you do $x or soft reset it does nothing. Disconnecting then connecting again will let you back but the jog controls are disabled. You have to restart the program to remedy this. 
[image]

jahnj0584, 2017-09-11:
yep, this is definitely just an issue if you send a file as $c and it has an error 33 or 20 (or probably any).
The file never finishes because the error so the GUI always thinks its still streaming a file.

winder, 2017-09-11:
I noticed this the other day, grbl is in `Check` mode which I don't handle properly when the stream is paused due to an error.

jahnj0584, 2017-09-11:
I recommend you make it cancel the stream after an alarm pop up. The
operator the. Fixes it and parses it again to see. If there's still a
mistake it'll be shown. Rinse and repeat until they give up or it parsed
properly 😅

jahnj0584, 2017-09-12:
Can confirm this happens on the Sep10 version

vMeph, 2017-09-12:
this bug always been present for a long time.
only solution is close UGS and restart program
would love to see this fix also...

breiler, 2018-03-19:
I've done some digging.

I added the communicator to the diagnostics panel to get some more information. And reproduce the problem:

1. Home my machine
2. Reset to Zero
3. Open circle.gcode that comes with the test files for UGS
4. Use "Check Mode"
5. Run the file and get the following error and close it (on my test rig this is a correct error because the soft limits will be hit if I try to go any higher on Z than 0):
[image]
6. Click "Soft reset" as the message in the console tells me
7. Click "Unlock" because we are still in an error state, and nothing happens. The console doesn't show that any commands are sent.
8. Refresh the current state in my custom UGS Diagnostics and can see that the communicator is still paused even though I've done a soft reset.
[image]

So if I add the reset of pause-flag in the BufferedCommunicator I can now do a soft reset and continue:
```java 
    @Override
    public void softReset() {
        this.commandBuffer.clear();
        this.activeCommandList.clear();
        this.sentBufferSize = 0;
        this.sendPaused = false; // Add a reset of the pause flag
    }
```

Is this an expected behaviour?

winder, 2018-03-20:
@breiler I think this is only a symptom of the problem. Errors like an unsupported gcode also cause streaming pauses but in this case, a reset shouldn't be required.

The way I'm handling errors in UGS is not ideal. The process goes:
1) Simple response parser in the Communicator class detects the error and pauses the communicator.
2) GUIBackend also detects that there was an error and must assume that the communicator has paused.
3) The GUIBackend must now update the control state accordingly. I'm not sure why I put the detection in GUIBackend instead of the Controller.

The 3rd step there doesn't work properly in check mode. Form your screenshot you can see the communicator is paused but the controller is not. The main reason for this is that UGS depends on GRBL to indicate that it is sending/idle/etc, and those states aren't updated in check mode.

There are a few ways I know this could be improved:
1) Add a `ControlState.COMM_CHECK` and use it in the spaghetti code to handle this chain of events. The code handling the error pause is not great, it starts in [BufferedCommunicator](https://github.com/winder/Universal-G-Code-Sender/blob/22d08af5bf3c2fb261c4b007e100236439ba58c6/ugs-core/src/com/willwinder/universalgcodesender/BufferedCommunicator.java#L304) where the streaming is paused (done at a low level to prevent sending additional commands after the error). The pause must then be detected in several other places places like [GrblController:rawResponseHandler](https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/GrblController.java#L170), [AbstractController:commandComplete](https://github.com/winder/Universal-G-Code-Sender/blob/22d08af5bf3c2fb261c4b007e100236439ba58c6/ugs-core/src/com/willwinder/universalgcodesender/AbstractController.java#L784),  [GUIBackend:commandComplete](https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/model/GUIBackend.java#L722), and possibly some other places.
2) Add a `communicatorPaused` event to `SerialCommunicatorListener` or redefine `void rawResponseListener(String response);` to `void rawResponseListener(String response, Boolean paused);`. With either of these changes, the communicator pause can be handled more directly in the controller.

breiler, 2018-03-20:
> Errors like an unsupported gcode also cause streaming pauses but in this case, a reset shouldn't be required.

I'm a bit confused in the error with soft limits that I provoked. GRBL is telling me that a soft reset must be done. Should the controller be resumed from the paused state before that reset-command could be sent?

winder, 2018-03-20:
GRBL has different classes of errors. When a soft limit is hit, GRBL has decided that the state is no longer valid and it cannot recover. So you are correct that GRBL must be reset to continue. This case isn't really handled by UGS in non-check mode either, you can keep trying to jog and you'll keep getting error popups.

Then there are the less severe errors like a parsing error, or an invalid gcode command. In those cases, GRBL can recover by skipping the command. With many workflows, this is a common occurrence, for example, one of the CAM programs I use puts in a tool change command (`M6 T102`), which causes an error from GRBL. After reviewing the error I'll click the play button and resume the program.

breiler, 2018-03-20:
Just so I understand. =)

So by always pausing the Communicator and let the Controller know, we could let the Controller  decide the severity of the error. Some would require a reset, others could be handled as in the skip command.

winder, 2018-03-20:
This is a tricky one. Here is some background about the error handline: [Grbl-v1.1-Interface#g-code-error-handling]( https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#g-code-error-handling)

The controller doesn't actually have an opinion right now. It is just updating its state according to whether or not the communicator is paused. To make matters more complicated, in some cases the communicator pauses even when it doesn't need to (i.e. the last command in a program, or a manual command).

The main idea is that when an error happens, the program halts and the user should be given enough context to decide what happens next.

breiler, 2018-03-20:
I'll have a look at it trying to create a new paused event from the communicator like you described.

However, my PR is still correct right? A soft reset should reset the internal state of the communicator?

winder, 2018-03-20:
I think you're right, I'll merge in that PR

breiler, 2018-03-21:
So I've taken a little dive into this. 

I think that when using check-mode it's best to use single stepping. I figure that because the controller is faster than the response handling, when an error occur it's already started sending the next command. When we handle the error by pausing the machine it starts reporting all errors of the messages that has already been streamed to the controller.

I think a part of the solution is that when we are starting a stream in check mode the controller should switch to single stepping, and when exiting to normal mode it should use the step mode selected in the settings.

I'm wondering about the enum `ControlState`, if feels like this is more of a `CommunicationState` and is quite clean as it is. I believe what we are missing is a ControllerState which should have all the states the application GUI should handle: JOG, RUN, HOLD, DOOR, ALARM, CHECK etc. This way it's easier for implementors of other controllers to know what kinds of states they need to support. In our case we would know that in the state CHECK this specific type and version of controller would need single stepping.

Other areas of the application would benefit from this. We have a couple of places where we look at the state string to identify which state the controller is in. For instance `TinyGController:getStateAsString` which has its own set that aren't compatible with the `MachineStatusPanel`.

winder, 2018-03-21:
I agree with all of this. Having `HOLD`/`ALARM` in particular would simplify a lot of brittle code which checks for things like `COMM_IDLE && isSending`. A change like this would probably be at a high risk for regressions, so it would need to be done incrementally. It might be possible to completely remove flags like `isSending` and `isPaused`.

Another complexity is that UGS still has support for older versions of GRBL which manages the GUI state manually. This mode is toggled with `handlesAllStateChangeEvents()`:
```java
    /**
     * Indicator to abstract GUIBackend implementation that the contract class
     * will handle ALL state change events. When this returns true it means
     * things like completing the final command in a stream will not
     * automatically re-enable buttons.
     */
    Boolean handlesAllStateChangeEvents();
```

breiler, 2018-03-24:
I've tested it with GRBL 0.9j and the problem seems to be fixed there as well.

breiler, 2018-03-26:
@jahnj0584 the changes should now be available in the nightly build. If you have the time I would really appreciate if you could verify that the issue is gone.

jahnj0584, 2018-03-27:
ill check this tonight after i finish this job.

breiler, 2018-04-14:
I'm closing this as I'm no longer able to reproduce it.

closed 2018-04-14 by breiler