winder/Universal-G-Code-Sender #761: Dynamic windows?

jahnj0584, 2017-09-11:
Is it within netbeans power to enable some dynamic windows or tools? For instance, when a file is sent, the jog controller is changed to the send progress window and the common actions toolbar could become the override window. 

Just a thought

winder, 2017-09-11:
I was looking into this a little bit after your comment on the advanced probing plugin. There is a feature called "roles" that should be able to do this.

In the future, I want to have a "gcode editor" role as well. Different roles for disconnected/connected/streaming might make sense too.

jahnj0584, 2017-09-12:
Just saw the new connection toolbar. I feel like that's good proof that what I'm asking for is possible if you so desire :)

closed 2017-11-05 by winder