winder/Universal-G-Code-Sender #763: Z height not changing during cuts

SirBasilton, 2017-09-16:
I've recently started having an issue with all versions of UGS (on the same computer) where the Z height does not change. If I'm cutting a 2d shape for example, it will do multiple cuts at the same z height rather than progressively lowering.

I've installed UGS on another pc and the issue is not there so is there a location where UGS stores data that I can wipe and see if that fixes my problem? Probing height data is stored somewhere but I cant figure out where it all is?

jahnj0584, 2017-09-16:
It's in my documents. Hidden folder called .ugs

winder, 2017-09-19:
@SirBasilton did resetting your settings fix the problem?

SirBasilton, 2017-09-20:
Sorry for the late reply. Yes, I managed to find the appdata folder and deleted everything in there. Problem solved!

closed 2017-09-20 by SirBasilton