winder/Universal-G-Code-Sender #764: "view gcode" window

jahnj0584, 2017-09-16:
Would be nice to have a tab beside the console to view the Gcode file that is loaded and allow for some changes manually or with a macro (change feed/spindle speed, rotate, scale, etc kinda like GCode ripper. First two are most valuable) and then auto saves it. You can't edit a file that's loaded already so you have to open a dummy file to release it, then reload just because you changed one line

vMeph, 2017-09-16:
this could be very good specialy if add a visualisaion simulation

jahnj0584, 2017-09-16:
Good idea. The line that is selected could glow a different colour in the
visualizer!

winder, 2017-09-16:
It can already do that, but the code is really fragile so it usually doesn't work. The GCode editor is pretty low priority right now -- I'm trying to focus on 2.0 milestone tasks.

[image]

jahnj0584, 2017-09-16:
Awesome!

vMeph, 2017-09-18:
@winder what is 2.0 milestone?

winder, 2017-09-18:
I tagged a bunch of tickets that I want to finish before having the first official 2.0 release. https://github.com/winder/Universal-G-Code-Sender/issues?q=is%3Aopen+is%3Aissue+milestone%3A2.0

winder, 2017-10-13:
Going to close this for now, it isn't going to happen anytime soon.

closed 2017-10-13 by winder