winder/Universal-G-Code-Sender #765: Visualizer VS reality differences

jahnj0584, 2017-09-16:
An issue I have with some files is that lines are greyed out before the machine actually gets to them. Should/could the visualizer be edited to only consider lines at and behind the tool tip are "sent" ?

winder, 2017-09-16:
Lines are sent to GRBL ahead of time, that's what the gray means. If the lines are long linear movements that could mean they are grayed out pretty far in advance. You're probably right, that distinction isn't really useful to most people.

Highlighting the line segment currently being executed might be interesting.

jahnj0584, 2017-09-16:
Ideally itd be both. Orange line is where the tool is currently
on/following and grey is whatever it has already passed and cut.

Would that be too hard to do?

winder, 2017-10-13:
It would be trickier than you'd think, I doubt I'll ever work on this.

closed 2017-10-13 by winder