winder/Universal-G-Code-Sender #766: Sender stop working

inFamousMcGrath, 2017-09-17:
Hi, i'm using last nighty build of UGS Platform, and i have Arduino Mega 2560 with GRBL 1.1f.
I'm working with a large G-code files(aroud 1-2 million rows), which have onli G1, G0 and M3/M5 commends. So here is a problem - sender randomly stops to send a rows, without any messages. It shows only on big Gcode files (current file size is 25 MB). GRBL status in verbose is Idle, spindle don't disable, so it just stay on last XYZ position with enabled spindle)). 
p.s. Sorry for bad English.

inFamousMcGrath, 2017-09-17:
Update - it happened only on notebook with 1 GB ram and Atom n450. i7 6700 +8Gb ram works fine.

winder, 2017-09-18:
@inFamousMcGrath try closing the console window, if that doesn't work also close the visualizer. If one or both of those fixes it I can investigate why they're leaking memory over time.

inFamousMcGrath, 2017-09-18:
I'll try, but there more bad news by this hour - it's stopped even on i7 6700 - after 6 hours, but stopped. I cant stop sending, it dont react to my actions, only restart helps. There 111000 rows, that less that previous file, which was around 1000000 rows. Visualizer disabled, but console not, i'll try to disable it.
[image]

vMeph, 2017-09-18:
that pendant sender have alot bugs, 
wish someone would fix,would be a great app for the smart phone

winder, 2017-09-19:
The pendant is another variable I wasn't even considering. When I wrap up with the features I'm working on now I'll try profiling the code to find where the memory is leaking. Until then if you leave the pendant disabled along with the console and visualizer hopefully the problem goes away. If it doesn't there is a much bigger problem than I realized.

inFamousMcGrath, 2017-09-19:
So, turning off the console doesn't helps me. It's strange, it happens only at night time, when i left my work))
Well, i have some ideas, what it will be:
Sender does not receive exact number of "ok" answers from GRBL (or it's an UGS bug), which need to continue sending. If there no error messages from UGS - i thought it really main option. I reccomend to synchronize UGS <ok> counter with verbose - GRBL 1.1 response number of queued motion anb avaliable buffer. Looks like Candle use right this method with GRBL 1.1.

Siand, 2017-10-12:
@inFamousMcGrath I've read some of the code base for the UGS and the reveiving of responses happens through events, and there is no frequency of checks there - the code triggers by receiving a message.

scratchboom, 2019-06-16:
@Siand so do you think it's GRBL problem?

breiler, 2020-04-01:
I've made some changes in the visualizer that seems to have a big impact on the performance. Please reopen if the problem still exists.

closed 2020-04-01 by breiler