winder/Universal-G-Code-Sender #767: Soft reset on connection

inFamousMcGrath, 2017-09-17:
Please add an option to disable soft-reset on connect. I really love new ugs-platform release, but it really have a lot of bugs, which need sometimes to restart UGS, so, when i restart and reconnect, program do a soft reset, and GRBL lost it position, and it will be impossible to done the job - i dont have limit switches.

jahnj0584, 2017-09-17:
You could try using your hard limits as a reference

<https://github.com/winder/Universal-G-Code-Sender/issues/767>, or mute the
thread

pelrun, 2017-09-17:
That's generally not UGS's fault. Arduinos are designed to reset on connect, as it's required to be able to reprogram them. Disabling it will require modifying the hardware.

inFamousMcGrath, 2017-09-17:
This is exactly UGS fault. I create my own C# simple sender, which connect to Grbl withoun any reset. And it's works fine. My Mega 2560 powered with exernal power supply, so it's independent from USB connection generally. So i can connect laptop, PC, or Android via bluetooth module HC 05 withoun any lost of coordinates.

winder, 2017-09-17:
Are you using G92 for offsets? Try using work coordinates instead, a soft reset shouldn't reset them.

It's true, UGS sends Ctrl-X on startup. Unfortunately, this is required right now to force GRBL to send it's version. UGS uses that to enable/disable features for backwards compatibility. It might be possible to make all that manually configurable instead, but I don't have plans to do that.

pelrun, 2017-09-17:
Ah. It's because you're connected via a bluetooth module that you're avoiding the connection resets. If you were connected via USB serial your "simple sender" would also be hard resetting the Arduino.

That's independent of UGS sending an explicit soft-reset, though.

inFamousMcGrath, 2017-09-17:
No, you wrong. Grbl with external power (no USB), will never need soft reset on connection, no matter, USB or bluetooth.  Its RECOMENDED by Grbl author, but not needed instead. Try it by yourself. You can connect to Grbl (with external powered Arduino) with any COM terminal program from any device - there will be no reset, unteel you send $X manually. Even more, GRBL 1.1 dont need reset, to see it's version - it send it when you connect. I dont know changes in Grbl Mega except  buffer and planner size, so it works on Uno, Nano etc.

inFamousMcGrath, 2017-09-17:
[image]
I dont think, that is simple sender, but... I works a lot of time with professional CNC, like Fanuc, Mitsubishi, Haas, etc. I wondering, that Grbl, which support a lot of codes, dont have a GUI, tah can use it at 100%.

winder, 2017-09-17:
@pelrun it is also possible to connect directly to the arduino rxtx pins to avoid the hardware connect reset. I think it's a pretty uncommon use case, but that's why UGS needs to send the soft reset.

pelrun, 2017-09-17:
I know, that's why I said it's because of the bluetooth module that he's not getting hard resets - they only ever have rx and tx connected.

Since it's only through sheer luck (or deliberate setup) that you can avoid the hard resets, I don't think it's that big a deal for UGS to send a soft reset. Fixing the bugs that require a restart of UGS would be the real solution to OP's problem.

jahnj0584, 2017-09-17:
I for one enjoy the auto $x. Makes life simpler

closed 2017-09-18 by winder