winder/Universal-G-Code-Sender #768: Command Counter not reset

BusbyGT, 2017-09-19:
I ran multiple Gcode files and noticed after basically each file, I ran into issue #593. But what I also noticed was that each file had a "Skipping command" console message. The number for the command never reset so each file just increased it. As you can see in this image, the Gcode file has 5599 lines (but only sends 5598) but somehow it skipped command 17,131.

OS: Raspbian
GRBL v1.f
Sept 10, 2017 nightly Platform

[image]

winder, 2017-09-19:
Thanks, I rarely run more than one program in a row so I would have never noticed this.

BusbyGT, 2017-09-19:
I'm assuming the skipped command that I'm seeing is the "M30" in my file even though the settings say to remove it. I'll test out that later.

winder, 2017-09-19:
Yeah, things like the M30 being removed, blank lines, or only comments would create this message.

winder, 2017-09-28:
I finally remembered the reason for this behavior. In older versions of UGS there was a table view of the commands as they were processed. In order for the table to display everything sent (manual, streaming, etc), every command needed to have a unique number so that they could be indexed into the command table.

I'd like to leave the command number alone, but I think this message can be improved. I've changed it to display some information about the command which was skipped:
```
Skipping line: M30
Skipping comment-only line: (Commands are very small, many as short as 7 characters.)
Skipping comment-only line: (This causes the buffer to fill very close to capacity.)
Skipping blank line #7
Skipping comment-only line: (Simple manual G-Code for an XY ruler)
Skipping comment-only line: (plate 50x50)
Skipping comment-only line: (origin is in left south)
Skipping blank line #11
Skipping comment-only line: (~~~setup~~~)
>>> G21
>>> G90
>>> G0Z5
Skipping blank line #16
Skipping comment-only line: (~~~ x ruler ~~~)
Skipping blank line #18
>>> G0X0.Y0.
>>> G1Z0.
>>> G1X0.Y5.
>>> G0Z5.
Skipping blank line #23
```

closed 2017-09-28 by winder

BusbyGT, 2017-09-28:
Awesome. That makes it easier to understand what is skipped and whether it matters.

Thanks.

ejs-ejs, 2018-10-25:
Why are you skipping M30?
I ran into it when using Autodesk Fusion360 GRBl/grbl postprocessor. It says `writeBlock(mFormat.format(30)); // stop program, spindle stop, coolant off` and no M5 commands. So i was left with the spindle running after the cycle ended.
Right now i've changed the postprocessor to
`writeBlock(mFormat.format(5)); // spindle stop
  writeBlock(mFormat.format(30)); // stop program, spindle stop, coolant off`

And it seem like "T" commands are stopping the cycle. Not still sure if it is UGS or grbl issue.

commit 4487e573 (2017-09-28): Improve skipped command message. Closes #768