winder/Universal-G-Code-Sender #769: Some G38.2 Command Ignored

BusbyGT, 2017-09-19:
Simple 3 axis probe script. When loaded, the file isn't fully loaded in the UI (no popup on Platform indicating an issue but the "Rows in File" stays at 0) but when streamed anyways, it stops before executing the "G38.2 Z-1 F2" command. The other 2 G38.2 commands are run but the third isn't. I want to say there is an error popup but I can't remember offhand.

Rpi 3 w/ Raspbian
GRBL v1.1f
Sept 10, 2017 nightly

[Kodiak-129644.txt](https://github.com/winder/Universal-G-Code-Sender/files/1312895/Kodiak-129644.txt)

BusbyGT, 2017-09-19:
The original description/image in #727 describes what I am seeing if I recall correctly. Notice how in the image it barfs on the third G38.2 command? I think that is the same error that I get when streaming it as well.

winder, 2017-09-19:
This is a bug, I agree it is a duplicate of #727

closed 2017-09-19 by winder

BusbyGT, 2017-09-20:
@winder I did more investigation tonight and the issue is actually the script that I attached in the first post has an error with a G0 command during parsing. The third G38.2 command is never actually sent to GRBL or loaded by UGS. I did figure out the issue with the script though.

The issue is that there is a space in the "G91 G0 Y- 0.625" command. If I remove the space, it loads perfectly fine but I haven't tested sending the script yet to the machine.

I've seemingly always had the space there and it has never caused a problem with GRBL. This must be an artifact of the recent change to the Gcode parsing, Is it a legal command?

[image]

Here's the console when you stream the file anyways:

[image]