winder/Universal-G-Code-Sender #771: Error when opening GCode file.

StuartB4, 2017-09-20:
Can't understand why I get this error when opening a gcode file. The file works perfectly in UGS 1.0.9, LaserGRBL, Candle and GRBL Panel.
[image]

winder, 2017-09-20:
Sorry about that, a regression snuck in recently while I was working on some parser improvements. I should have a fix available in the next couple days.

rmisko11, 2017-09-20:
My understanding is that G1 is a move command, G1 F9000 isn't a move.
Maybe change G1 F9000 to G1 X0 F9000 or just delete the G1 part of G1 F9000?

winder, 2017-09-20:
More precisely: http://linuxcnc.org/docs/html/gcode/overview.html#_modal_groups

And even more precisely: http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g0
"For rapid motion, program G0 axes, where all the axis words are optional."

I made a lot of fixes and improvements around actual spec compatibility over the past few days but am still testing.

StuartB4, 2017-09-20:
It seems to have fixed it'self. I updated Java and it's working. I have another problem though, nothing shows up in the visualizer window when I l oad a gcode file in.

StuartB4, 2017-09-20:
It seems to have fixed it'self. I updated Java and now it's ok. I have another problem though.
When I load a gcode file in, nothing shows up in the visualizer window.

chamnit, 2017-09-20:
FYI. `G1` by itself is valid, just as long as the feed rate has been previously set. `G1 Fxxx` and `G0` are also valid. These are all non-moves and are usually used to set up the modal state by some CAM programs.

winder, 2017-09-20:
@chamnit thanks. My parser has always been pretty fast and loose with the standard, in most cases just looking at the axis words and connecting the dots is pretty convincing for drawing a representation of the program. But now I'm trying to align it a bit closer to the spec so that gcode transformations can be made with more confidence.

rmisko11, 2017-09-21:
I am curious as to the name of the standard that defines G-Code?

The only I found, although I didn't look very hard, is:

 RS274/NGC Interpreter which for both G0 and G1 states:

It is an error if:
• all axis words are omitted

winder, 2017-09-21:
@rmisko11 there are a lot of people with very strong opinions on that subject. My general understanding is that the spec GRBL uses the interpretation by LinuxCNC, so that's the one I've been using for no other reason (I haven't really looked at the others).

Looking at that RS274/NGC spec for the first time, I can see why sonny prefers the linuxcnc version.

Check this out in a different section:
```
In most cases, if axis words (any or all of X…, Y…, Z…, A…, B…, C…) are given, they specify
a destination point. Axis numbers are in the currently active coordinate system, unless explicitly
described as being in the absolute coordinate system. Where axis words are optional, any omitted
axes will have their current value.
```
So G0 is invalid if axis words are omitted.... but axis words are optional?

rmisko11, 2017-09-21:
I agree the little I read didn't seem clear to me and thanks I will look at
linuxCNC to satisfy some of my curiosity.

dnpalmer, 2017-09-23:
I ran into a similar, but slightly different error. 

[image]

Had the following code:

G90
G20
G17 G64 P0.001 M3 S3000
F5.00
G0 Z0.2500
G0 X0.1250 Y3.5100
G1 Z-0.0600
G1 X0.1250 Y3.5100
G1 Y6.8696
G1 X0.1404 Y6.8750
G1 X6.8696
G1 X6.8750 Y6.8596
G1 Y0.1304
G1 X6.8596 Y0.1250
G1 X0.1304
G1 X0.1250 Y0.1404
G1 Y3.5100
G0 Z0.2500
M5
M2

pelrun, 2017-09-24:
> So G0 is invalid if axis words are omitted.... but axis words are optional?

Basically means that it's mandatory to have at least one axis word, but it doesn't matter which one it is.

winder, 2017-09-27:
This issue should be resolved with the latest nightly build.

closed 2017-09-28 by winder