winder/Universal-G-Code-Sender #778: Probe plugin will not allow negative pulloff

alexcampbelledm, 2017-10-12:
In the October 8th release, there is a bug in the probe plugin that will not allow a correct pulloff in the positive Z direction. If you probe Z positive or negative, it pulls off in the negative Z causing a crash into the probe plate

winder, 2017-10-12:
@alexcampbelledm the `retractHeight` setting is used for the final offset, you can change that setting to a negative number as a temporary workaround.

Could you describe your use case?

Maybe UGS could detect if Z is using a negative offset and skip the final retract in such a case (or do a small one instead).

alexcampbelledm, 2017-10-13:
Probe distance direction is set to -10. I am unable to set a negative retract height. In fact I am unable to go less than 10mm. This makes it attempt to retact in a negative direction. The X and Y probes function as expected. 

My Z axis is negative down, so I would think that a positive retract should move the bit up, but it does not. Do you need the code it is generating?

alexcampbelledm, 2017-10-13:
@winder 
Compared the G-Code generated and it seems the problem is the pulloff command is G90, not a G91. It zeros my probe height at 19.52, but then it issues a g90, which sends to the absolute distance of 10mm, which is below my probe. I found that if I want the 1mm pulloff, setting the retract amount to 20.52 gives my desired outcome.
In your response, you used the variable "retract height", but in the program it is labeled "Retract Amount." Is this maybe the issue? The "amount" would be contiguous with previous versions, while specifying a height would be a new method. 

Thanks for your time on this!

winder, 2017-10-25:
@alexcampbelledm I found a bug in the retract which should be fixed now. I also added a check for probing in the positive direction which will skip the final retract.

alexcampbelledm, 2017-10-25:
Thanks. I am excited to give it a try!

closed 2017-11-05 by winder