winder/Universal-G-Code-Sender #779: Jog Module

winder, 2017-10-13:
To work around some known issues with keybinding in netbeans platform, a special jog module would be useful. Initially discussed [here](https://github.com/winder/Universal-G-Code-Sender/issues/709#issuecomment-324061451)

What I would like to do is have a special "JOG MODE" button in the toolbar which would create a modal dialog. Jogging would be disabled unless that dialog was open, this way there would be no chance for accidental jogs (or ignored jogs). Something like this is what I have in mind where the text boxes let you reconfigure (or see) the bindings right away:
[image]

This would also allow me to implement more advanced jogging where I do things like start jogging on KEY_PRESSED and stop on KEY_RELEASED and use the [joystick implementation described here](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging#joystick-implementation).

vMeph, 2017-10-13:
been waiting for this to happen, make the UP and DN arrowy keys to work and jogging on key press and release  configurations
just dont think that would be a good idea to only work if a dialog box is open
 would be good, it should just have a place to set  a 1 time keys,steps  just like the pic above etc.. then you just use has normal
just was wondering if there could be a place to set the ms seconds for the keys to respond, i say this cause i have used before a joystick program that allows to set the control ms seconds for responses of moves that would make a good sync on key press and key release, cause the way it was before if you keep your key press then when you release it will keep moving until buffer is empty

jahnj0584, 2017-10-13:
Yea, i wouldn't make it such a hard requirement. I actually have no issues
with how it is now, seeing as i use CTRL as a trigger to activate any
keyboard commands. 0-1-2-3 on the numpad are my jog sizes (.01,.1,1,10) and
* for when i want to do 100mm.

I keep the speeds at 7000.

I'd rather remove the machine state ticks from the DRO and make them a
seperate window option.

winder, 2017-10-13:
@jahnj0584 that the status tick change is part of #739

The existing jog functionality won't be changing, this idea is for a new plugin for slicker joystick style maneuvering of your machine.

breiler, 2023-02-27:
I am closing this as the jog shortcuts should now work properly as of #2163 and I don't think there is a need for a separate plugin.

closed 2023-02-27 by breiler