winder/Universal-G-Code-Sender #781: when a sending gcode to arudino i got this error

electroblazze, 2017-10-14:
[verbose]<Idle,MPos:0.000,0.000,0.000,WPos:50.218,-9.449,0.178>
[verbose]<Idle,MPos:0.000,0.000,0.000,WPos:50.218,-9.449,0.178>
[verbose]<Idle,MPos:0.000,0.000,0.000,WPos:50.218,-9.449,0.178>
[verbose]<Idle,MPos:0.000,0.000,0.000,WPos:50.218,-9.449,0.178>
Skipping command #0
Skipping command #1
>>> G21 (metric ftw)
>>> G90 (absolute mode)
>>> G92 X0.00 Y0.00 Z0.00 (you are here)
Skipping command #5
>>> M300 S30 (pen down)
>>> G4 P150 (wait 150ms)
ok
ok
>>> M300 S50 (pen up)
>>> G4 P150 (wait 150ms)
Skipping command #13
Skipping command #15
Skipping command #48
Skipping command #49
Skipping command #61
Skipping command #62

winder, 2017-10-14:
@electroblazze what is the error? What do you expect to happen?

closed 2017-10-25 by winder

suganth789, 2018-02-12:
here p150 means (150 seconds delay not 150 milli second) so change this to p03(3 seconds delay) or whatever you like

gwag18, 2018-05-22:
hi.
what's the matter her g-code?

M300 S50
G90
G21
G1 F2000
G1  X5.0397 Y5.0034
M300 S30
G4 P1
G1 F2000.000000
G1  X5.0866 Y4.7026 F2000.000000
G2 X4.8297 Y4.581 I-0.2996 J0.3007
G2 X4.307 Y4.8569 I-0.0537 J0.5313
G2 X4.2081 Y5.43 I0.7931 J0.4319
G2 X4.7664 Y6.1841 I1.003 J-0.1588
G2 X5.6696 Y6.2705 I0.5671 J-1.1644
G2 X6.6371 Y5.4342 I-0.3812 J-1.4186
G2 X6.7028 Y4.1501 I-1.6916 J-0.7303
G2 X5.5963 Y2.9343 I-1.8681 J0.5887
G2 X3.9897 Y2.8914 I-0.8581 J2.0331
G2 X2.563 Y4.2778 I0.8182 J2.2693
G2 X2.545 Y6.2833 I2.6093 J1.0263
G2 X4.1959 Y7.967 I2.7167 J-1.0126
G2 X6.5096 Y7.9601 I1.1482 J-2.9046
G2 X8.3988 Y6.0254 I-1.2517 J-3.112
G2 X8.3659 Y3.2967 I-3.5267 J-1.322
G2 X6.1715 Y1.1433 I-3.5608 J1.4337
G2 X3.1498 Y1.2019 I-1.4382 J3.7756
G2 X0.7969 Y3.6842 I1.6836 J3.952
G2 X0.8819 Y7.1366 I4.4433 J1.6178
G2 X3.6194 Y9.7607 I4.4031 J-1.8534
G2 X7.3495 Y9.6496 I1.7284 J-4.6462
G2 X10.1667 Y6.6198 I-2.1146 J-4.7908
G2 X10.029 Y2.4434 I-5.3595 J-1.9137
M300 S50
G4 P1
G1 F2000
G1 X0 Y0