winder/Universal-G-Code-Sender #782: Stop and Pause button are greyed out.

Chromium-x, 2017-10-16:
After i start a job the buttons for stop and pause (menu bar) are still grayed out.
The buttons in the drop down menu working properly.

UGS Platform (Version 2.0 [nightly] / Oct 15, 2017)

winder, 2017-10-16:
What version of GRBL are you using?

RoadkillUK, 2017-10-16:
I had the same with mine yesterday, couldn't stop or pause also the 'Send Progress' wasn't updating.

grbl 1.1
WinXP
1G mem

however I have a spare grbl board for testing and that does the same on the same PC, but if I use my main PC (not in the same room as the machine) it's fine.

grbl 1.0
Win10
8G mem

vMeph, 2017-10-16:
Same here stop button dont works
Grbl 1.1

jahnj0584, 2017-10-16:
I have 1.1 and the latest stable version. The buttons are fine on my end.

winder, 2017-10-17:
Thanks for the reports, for some reason this doesn't happen when I run from my editor, I think it should be resolved now.

Chromium-x, 2017-10-17:
It works.

Thank's.

closed 2017-10-17 by Chromium-x