winder/Universal-G-Code-Sender #783: NullPointerException

RoboDad, 2017-10-16:
Hi, when I open a file with the following g-code sequence, I get an exception (see below):

G00 G17 G20 G40 G49 G80 G90

This is basically a preamble sequence that my CAM program (MasterCAM) is generating. 

I'm using UniversalGcodeSender-v2.0.0-stable

On a somewhat related note...  Does anyone know of a MasterCAM post for GRBL/UGS?

Thanks,

Geoff

Oct 15, 2017 8:44:34 PM com.willwinder.universalgcodesender.MainWindow browseButtonActionPerformed
SEVERE: Problem while browsing.
java.lang.NullPointerException
        at javax.vecmath.Tuple3d.<init>(Tuple3d.java:91)
        at javax.vecmath.Point3d.<init>(Point3d.java:74)
        at com.willwinder.universalgcodesender.types.PointSegment.<init>(PointSegment.java:82)
        at com.willwinder.universalgcodesender.gcode.GcodeParser.addLinearPointSegment(GcodeParser.java:232)
        at com.willwinder.universalgcodesender.gcode.GcodeParser.handleGCode(GcodeParser.java:312)
        at com.willwinder.universalgcodesender.gcode.GcodeParser.processCommand(GcodeParser.java:210)
        at com.willwinder.universalgcodesender.gcode.GcodeParser.testState(GcodeParser.java:438)
        at com.willwinder.universalgcodesender.gcode.GcodeParser.preprocessCommand(GcodeParser.java:421)
        at com.willwinder.universalgcodesender.gcode.util.GcodeParserUtils.write(GcodeParserUtils.java:71)
        at com.willwinder.universalgcodesender.gcode.util.GcodeParserUtils.processAndExportText(GcodeParserUtils.java:124)
        at com.willwinder.universalgcodesender.gcode.util.GcodeParserUtils.processAndExport(GcodeParserUtils.java:54)
        at com.willwinder.universalgcodesender.model.GUIBackend.preprocessAndExportToFile(GUIBackend.java:163)
        at com.willwinder.universalgcodesender.model.GUIBackend.initializeProcessedLines(GUIBackend.java:909)
        at com.willwinder.universalgcodesender.model.GUIBackend.setGcodeFile(GUIBackend.java:476)
        at com.willwinder.universalgcodesender.MainWindow.browseButtonActionPerformed(MainWindow.java:1271)
        at com.willwinder.universalgcodesender.MainWindow.access$3000(MainWindow.java:84)
        at com.willwinder.universalgcodesender.MainWindow$28.actionPerformed(MainWindow.java:993)
        at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
        at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
        at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
        at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
        at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
        at java.awt.Component.processMouseEvent(Unknown Source)
        at javax.swing.JComponent.processMouseEvent(Unknown Source)
        at java.awt.Component.processEvent(Unknown Source)
        at java.awt.Container.processEvent(Unknown Source)
        at java.awt.Component.dispatchEventImpl(Unknown Source)
        at java.awt.Container.dispatchEventImpl(Unknown Source)
        at java.awt.Component.dispatchEvent(Unknown Source)
        at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
        at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
        at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
        at java.awt.Container.dispatchEventImpl(Unknown Source)
        at java.awt.Window.dispatchEventImpl(Unknown Source)
        at java.awt.Component.dispatchEvent(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)

Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
        at com.willwinder.universalgcodesender.utils.GUIHelpers.lambda$displayErrorDialog$0(GUIHelpers.java:45)
        at java.awt.event.InvocationEvent.dispatch(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)

winder, 2017-10-16:
What version are you using? When I open that code on the nightly build I get the error below. It looks like `G80` is supposed to cancel canned cycles, so it looks like the UGS error is wrong. GRBL and UGS don't support canned cycles though so there's no need for it at all.
[image]

vMeph, 2017-10-16:
He sounds like using wrong post processor ,needs a post processor that matchs grbl

RoboDad, 2017-10-16:
I'm using the July 28, 2017 Stable build of UGS, Classic GUI, Java 8 (latest), grbl-Mega (latest build), Windows 10

Does anyone know of an existing MC post processor for grbl?  That would be great.  I couldn't find one with a cursory search.

Assuming I can't find an existing post processor for grbl, I will modify my Mach4 post processor to work with grbl.  That's basically the route I was taking when I hit the exception in the original post.

Thanks again for any help.

RoadkillUK, 2017-10-16:
@RoboDad The PP is part of the software you are designing with .. MasterCam. (of which I have no experience)

RoboDad, 2017-10-17:
@winder I tried the nightly build and it works, as you described.  i.e., I get the error message instead of the NullPointerException.  So it's only the "stable" build that has the problem.  I'll just work with the nightly build that I pulled tonight.  Thanks!

winder, 2017-10-17:
@RoboDad thanks for the update

closed 2017-10-17 by winder