winder/Universal-G-Code-Sender #784: Easy calibrate feature

danielboone1, 2017-10-19:
Hey...first of all, I love your software. I never even dreamed I would be capable of building or running a CNC, but you made it super easy. I'm no genius so your work is a true asset. I'm very new to this and I was wondering if there were any plans to make a simple calibration tool on the GUI? What I mean is, a way to push a button and say X is at 0 here, and at 12 inches here. Same for Y and Z. Mach3 has a similar feature. This way, you could use a machine with different lead screws and motors without the headache of manipulating a lot of code. Well...maybe there is something like that and I haven't wrapped my mind around it yet? Anyways, thank you for all you do!

winder, 2017-10-19:
Could you describe what this tool would do in a little more detail? I'm trying to understand how it could be easier than jogging the machine back and forth with the existing buttons.

jahnj0584, 2017-10-21:
Will, he means an axis calibration plug in. I suggested this last year.
You'd basically perform the function of this but within UGS

https://nathan7.eu/stuff/RepRapCalculator/RepRapCalculator.html#MotorStuffPrecise

danielboone1, 2017-10-21:
Okay, sure! Thanks for the quick reply. I'm building a DIY CNC in my shop
and I am using motors and lead screws with slightly different specs on each
axis. So, first I guess maybe I should ask a question...how do I set the
distance? I know that I can jog the controllers to move X, Y, and Z and
zero them. But how does the machine know how far it has traveled in, not in
steps, but in distance units such as mm or in.? What I'm saying is, would
there be a way to add a feature where you could calibrate the distance (in
steps) to a known standard like a gage block? How do I tell the machine how
far it has traveled? If I am using different lead screws and motors, it
will throw off the linear travel in relation to the other motors. But if
there was a way to calibrate the machine to show  "On Z, from point A to
point B, the distance is 100mm or 12" etc. On X, this many steps=this
distance in (units). On Y, this many steps=this distance in (units). Does
that make sense? So far, the only way I have found to dial this in is by
manipulating the code. A simple tool that would say "Calibrate" and then
enter a known distance between two points in a text box, followed by a
prompt to set to zero, then move spindle to your set distance, say 100mm
away. Then you could simply hit "set calibratioin" and "save" and the
machine would now know how far it has traveled in units. Mach 3 has a
similar feature. Very handy for DIY machines or unconventional setups. If
there's already a way to do this, please let me know! I have combed online
and cannot figure it out!

On a side note...just a simple yes/no...have you heard of anyone using a
USB generic gamepad (SNES) to jog? Not sure how to map the keys from the
gamepad output and convert the output to UGS hotkeys.

Thanks again and keep up the good work!

danielboone1, 2017-10-21:
> Daniel read my comment and check the link. You start with the theoritcal
> steps/mm then use math to make it meet reality
> https://youtu.be/LRagjLewSJg
>
>
> I also used an Xbox controller. All you do is use a program to map the
> buttons to keyboard keys and edit your shortcuts

Thank you so much! That works fine for me. I got my SNES controller working
fine with that program. I will try to get the math all sorted out now.
Thanks much

pelrun, 2017-10-23:
You're better off calculating this rather than trying to measure movement distances. Stepper motor steps per revolution times stepper driver microsteps divided by leadscrew pitch (times number of leadscrew starts if it's multistart) gives you steps/mm directly.

jahnj0584, 2017-10-23:
James that only works in a perfect world, which we aren't. You always have
to calibrate manually too.

pelrun, 2017-10-23:
If it doesn't match up, then there's something wrong. It's not a "welp, reality doesn't match theory" issue, it's a "you're probably losing steps somewhere" one.

jahnj0584, 2017-10-23:
Wrong James. It means "welp, my ballcrew wasn't rolled to a perfect
thousandths over 50 inches"

Yes it's a start, but my 80steps is 80.7 and my 45 is 44.87. If you don't
think the difference matters, feel free to have things cut out 1/4" short.
I cut things to 0.001" or 0.01mm precision. Many of the hobbiests here
don't do the raw number will be fine since all they do is "artistic'
projects like key chains and 3D relief carvings. I make things that have to
perfectly mate to other perfect parts.

winder, 2017-10-23:
I don't have any problem with supporting both of these calibration methods in UGS. Execution-wise I think a basic calculator app (which possibly updated the $ settings a when you click apply) would be simpler than trying to coordinate any more complex calibration.

For example, in an ideal world, you'd probably be using a dial indicator to get things really well aligned, other folks might only have a vernier caliper to measure calibration parts. Automation to support all the different techniques doesn't seem like something I can do. So probably something like that RepRapCalculator (and maybe a built in calibration program) might be the best I can do for a calibration plugin built into UGS.

pelrun, 2017-10-23:
If your leadscrew isn't exactly 80, don't use 80 in the calculation! It's also the one thing that you can measure more precisely the longer your axis is. You can measure a test part afterwards and adjust based on that - either of these is going to be far better than trying to accurately measure the real distance of an axis movement to 0.01mm.

jahnj0584, 2017-10-23:
Jamie's you don't know what the value is until after you calibrate the
machine. I'm done discussing it.Do whatever works for you but my three
machines are all good this way.

Winder, the reprap calculator AND the auto adjust setting would be amazing.
I honestly don't expect anything more complex then that even in a perfect
world

vMeph, 2017-10-24:
Having a leadscrew and belt microstep calculator would be a very good option, there is a plication on playstore for smartphone called StepperMillimeter could be something like that to get the idea

danielboone1, 2017-10-24:
I appreciate all of the input. I still think a simple plugin would set the
software above the rest. Just saying...calibbrating to a known point of
origin would always be more effective. All the same, this definitely gets
me pointed in the right direction. Thank you so much to all of you!

winder, 2017-12-01:
[Moved to google group](https://groups.google.com/forum/#!topic/universal-gcode-sender/diYBTV9ss24)

closed 2017-12-01 by winder