winder/Universal-G-Code-Sender #786: Homing delay on a single axis

aforww, 2017-10-25:
I'm not even sure if this is a UGS issue but gotta start somewhere.

So I've installed a new CNC shield V3.51, updated GRBL to 1.1F and I'm running into an issue with my Z axis homing. I'm using hall effect sensors which have been great right up until now. THey have an integrated LED that indicates when it trips. 
The issue I'm having is there is a 2-3 second delay between when the switch trips (indicated by the led) and when UGS starts registering it as seen in verbose mode.

Anyone have any ideas? It's only the one axis that's doing it so I don't suspect the debounce delay settings.

winder, 2017-10-25:
GRBL doesn't respond to status messages while it's homing so that could look like a problem even though it's normal.

Is the delay in the way the machine moves or in the UGS display?

You mentioned that you weren't having any trouble earlier though -- what pieces changed with your setup?

What version of UGS are you running?

aforww, 2017-10-25:
The delay is in UGS (or GRBL?) actually seeing the switch as activated. Example:
I initiate the homing cycle the Z axis moves up and activates the switch and continues on, crashing into the top of the Axis, it continues seeking for about two seconds and finally recognizes the switch was activated.  X and Y both register immediately.

I switched out my board from V3 to V3.51 and updated GRBL to 1.1F I'm away from the computer at the moment but I believe the UGS-P version was a nightly from a week or so ago. 

I suspect it's a GRBL issue but I just don't know at this point.

winder, 2017-10-25:
Gotcha. One thing to double check is the wiring, definitely verify that the traces are correct: https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

Homing is completely managed by GRBL (UGS just sends a `$H` command to initiate the sequence), so there probably isn't anything in the software that could cause problems.

aforww, 2017-10-25:
> wiki/Wiring-Limit-Switches

Yea I kinda figured it's origin was GRBL but thought I'd probe here as
well. I posted the same issue in the GRBL github. I find you and most
people here are quicker to respond with feedback though. Lol. I'll check
the traces when I get home. I'm finding this v3.51 board to be quite the
hassle. All this so I don't have to turn on my VFD and turn a knob for
spindle speed.... Not sure it's worth it anymore lol.
Thanks for the help. Can go ahead and close this one out.

closed 2017-10-25 by winder

jahnj0584, 2017-10-25:
Afor you can control the spindle just with the arduino

aforww, 2017-10-25:
> issues/786>

I'm using a VFD and 1hp spindle so I have a relay for triggering the VFD to
on and a PWM to analog board for controlling speed. The V3.51 board was
just to already have the pins for those things available instead of hacking
the V3 board.

To make matters worse, I thought to myself that maybe my Z axis issue was a
poor connection at the sensor. Went to unplug the wires from it to reseat
everything and managed to jerk the plug and entire trace right off the
board... This build is testing my resolve. Lol. However in moving homing
switches around things started triggering correctly minus one switch. Think
all these little things are effecting my concentration as you would think I
would test the Z issue with a different switch before going off asking
questions thinking it's software related. Or maybe I'm just going senial.

Either way, good on ya for being quick to reply. As my post on the GRBL
github has still gone unanswered. Even though it may be moot.