winder/Universal-G-Code-Sender #787: Hard limits double-triggering with $21=1

jahnj0584, 2017-10-25:
First you jog too far and hit the switch. You reset and unlock, and try to jog away but when the switch releases it also again triggers the hard limits. You must repeat the process to finally get away, then re-home the machine. This is unnecessary and frankly, makes $21 a setting most people would want to disable automatically which just leads to a safety issue, or a broken project.

winder, 2017-11-05:
This is controlled by GRBL, I don't think there's anything I can do in UGS to improve the situation.

My switches are simple lever switches so there's no way to go past them. I guess you are using some sort of magnetic sensor? Could you add more magnets so that the switch remains active until you get back to a safe location?

closed 2017-11-05 by winder

jahnj0584, 2017-11-05:
nay, lever style as well. Its just that GRBL will set a hard limit with the on and off action of the switch. I have disabled it for now anyways as certain RPM of my spindle cause a false hard limit. Ill shield them some day...