winder/Universal-G-Code-Sender #788: Using Universal GCode Sender and Arduino Uno to control a CNC with servomotors

Kheco, 2017-10-26:
Hi.

I am developing a little project building a cnc plasma but using servomotors instead of steppers.
Using the same signals from Arduino that UGS provides can I control the servos as the steppers. (Sorry if looks like stupid question).

I'll be updating the project status in case somebody had the same doubts.
Thank you! Best regards!

winder, 2017-10-26:
Are you developing your own firmware? UGS has a plugin system to support firmware other than GRBL.

rmisko11, 2017-10-26:
Is there a plugin to support Marlin?  Where can I find information on the
plugin system?

winder, 2017-10-26:
There could be, so far nobody has tried integrating another controller. I implemented TinyG to test out the system (Though it's pretty out of date now since nobody ended up using it). The documentation is here: http://winder.github.io/ugs_website/dev/backend_development/

rmisko11, 2017-10-26:
Thanks Will

Kheco, 2017-10-26:
No, I'm trying to use GRBL.
I have a Y00CNC-NT65-3X and it's working with Mach3 Software (But it's pretty out of date too), its most current firmware for Mach3 could be an AKZ-250. But i'd like to use UGS and Arduino better, the problem is that i'd use industrial Servomotors (Yamaha) instead of steppers.

winder, 2017-11-05:
@Kheco you'll have to investigate the GRBL firmware. There may be forks which can control servo motors, but it isn't a supported feature. If you do this you can use software like UGS (or others grbl UIs) with no changes.

closed 2017-11-05 by winder

Sailjunky, 2018-10-22:
Some servo drives have step and direction which you can interface directly to the pulse and direction outputs from the Arduino.  
You configure those drives in position mode and then they behaves pretty much like a stepper. except that they will not loose pulces. 
Unless there is an autoTuning function, you will also have to learn to tune in torque mode, then velocity mode, then position mode to get yoour system running smooth and precise.    
That way, it keeps the project much simpler compared to using the encoder feedback with DAC outputs. 
I didn't see any Arduino solution for multi axis servo control yet.