winder/Universal-G-Code-Sender #790: UGS Platform won't start (Win10, 32bit)

WePaHe, 2017-10-31:
Win10, 32bit, german, ugsplatform.exe 
HP nw8240, 2GB RAM, ATI FireGL5000

on double click to ugsplatform.exe
the UGS startup notice appears,
then there is a massage that UGS is not totally translated to my language (german).
Next appears a notice that it is a nightbuilt and 
the UGS desktop comes up for half a second
and then UGS terminate without any hint?

Java is Version 8 update 151.
Starting ugsplatform in windows kompatibilty modes and 
reducing the video mode to 1280x1024 and 256 or 65k colors bring no change.

The Classic UGS works as v1.09 and 2.0 Nightbuilt. 

At my Desktop PC (Win10-64bit)  both UGSplatform versions 32bit and 64bit works.

any Idea?

winder, 2017-11-01:
Do you know how to launch ugsplatform.exe from the command line? There should be an error message in the console if you run that way.

WePaHe, 2017-11-01:
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x27834e46, pid=6120, tid=0x0000077c
#
# JRE version: Java(TM) SE Runtime Environment (8.0_151-b12) (build 1.8.0_151-b12)
# Java VM: Java HotSpot(TM) Client VM (25.151-b12 mixed mode windows-x86 )
# Problematic frame:
# C  [atioglxx.dll+0x354e46]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# An error report file with more information is saved as:
# C:\Users\xxxxxxx\AppData\Local\Temp\\hs_err_pid6120.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.

[hs_err_pid6120.log](https://github.com/winder/Universal-G-Code-Sender/files/1434762/hs_err_pid6120.log)

WePaHe, 2017-11-01:
``
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x27834e46, pid=6120, tid=0x0000077c
#
# JRE version: Java(TM) SE Runtime Environment (8.0_151-b12) (build 1.8.0_151-b12)
# Java VM: Java HotSpot(TM) Client VM (25.151-b12 mixed mode windows-x86 )
# Problematic frame:
# C  [atioglxx.dll+0x354e46]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# An error report file with more information is saved as:
# C:\Users\Werner\AppData\Local\Temp\\hs_err_pid6120.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

the complete Log-File see below
[hs_err_pid6120.log](https://github.com/winder/Universal-G-Code-Sender/files/1434786/hs_err_pid6120.log)

WePaHe, 2017-11-01:
xbjhsbj

> A fatal error has been detected by the Java Runtime Environment:
EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x27834e46, pid=6120, tid=0x0000077c
JRE version: Java(TM) SE Runtime Environment (8.0_151-b12) (build 1.8.0_151-b12)
Java VM: Java HotSpot(TM) Client VM (25.151-b12 mixed mode windows-x86 )
Problematic frame:
C [atioglxx.dll+0x354e46]
Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
An error report file with more information is saved as:
C:\Users\Werner\AppData\Local\Temp\hs_err_pid6120.log
If you would like to submit a bug report, please visit:
http://bugreport.java.com/bugreport/crash.jsp
The crash happened outside the Java Virtual Machine in native code.
See problematic frame for where to report the bug.

the complete Log-File see below

[hs_err_pid6120.log](https://github.com/winder/Universal-G-Code-Sender/files/1434805/hs_err_pid6120.log)

winder, 2017-11-01:
Unfortunately it looks like this is crashing in the 3d library I'm using. My only guess is to try updating drivers for your graphic card.

WePaHe, 2017-11-01:
Thanks. I'll try to find another grafic card Driver.

WePaHe, 2017-11-04:
after changing the ATI display driver from V8.483 to 8.503 all works

closed 2017-11-04 by WePaHe

winder, 2017-11-05:
Thanks for the update!

OtoVel, 2017-11-26:
hello, I use UniversalGcodeSender.jar v1.0.9. and so far it has worked normally. Some time ago I installed Android Studio, which also wanted to enable Java virtual machine. today I tried to use UniversalGcodeSender.jar and fall when I click on OPEN. here's the error right, can you tell me what to do to make it work again?

Microsoft Windows [Version 10.0.16299.64]
(c) 2017 Microsoft Corporation. All rights reserved.

C:\Roman\Projekty\drevene vyrobky\_Embraser\UniversalGcodeSender-v1.0.9>java -jar -Xmx256m UniversalGcodeSender.jar
nov 26, 2017 5:15:38 PM com.willwinder.universalgcodesender.utils.SettingsFactory loadSettings
INFO: Loading settings.
nov 26, 2017 5:15:38 PM com.willwinder.universalgcodesender.utils.SettingsFactory loadSettings
INFO: Settings file location: C:\Users\roman\UniversalGcodeSender.json
nov 26, 2017 5:15:39 PM com.willwinder.universalgcodesender.model.GUIBackend applySettings
INFO: Applying settings.
nov 26, 2017 5:15:39 PM com.willwinder.universalgcodesender.model.GUIBackend addControllerListener
INFO: Adding controller listener.
nov 26, 2017 5:15:39 PM com.willwinder.universalgcodesender.model.GUIBackend addControlStateListener
INFO: Adding control state listener.
nov 26, 2017 5:15:41 PM com.willwinder.universalgcodesender.model.GUIBackend connect
INFO: Connecting to GRBL on port COM3
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=9760, tid=2672
#
# JRE version: Java(TM) SE Runtime Environment (9.0+11) (build 9.0.1+11)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (9.0.1+11, mixed mode, tiered, compressed oops, g1 gc, windows-amd64)
# Problematic frame:
# C  [jSSC-2.8_x86_64.dll+0xb5db]
#
# No core dump will be written. Minidumps are not enabled by default on client versions of Windows
#
# An error report file with more information is saved as:
# C:\Roman\Projekty\drevene vyrobky\_Embraser\UniversalGcodeSender-v1.0.9\hs_err_pid9760.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

winder, 2017-11-26:
Do you have android studio running at the same time? This is a problem in some native code used by the serial library UGS is using.

OtoVel, 2017-11-26:
hi, no, android studio does not run in same time. I just thought it might be a problem because it was a bigger JAVA installation and update. thanks for any advice

winder, 2017-11-26:
Sorry you're having so much trouble with your machine, I think you've hit a bug in about every native  library UGS uses.

If you copy this file `C:\Roman\Projekty\drevene vyrobky_Embraser\UniversalGcodeSender-v1.0.9\hs_err_pid9760.log` I can take a look, but there probably isn't much I can do.

OtoVel, 2017-11-26:
I checked windows updates, the first son installed the last one but still the same error, I checked the actuality of the graphical driver, windows talk from using the last

OtoVel, 2017-11-26:
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=9760, tid=2672
#
# JRE version: Java(TM) SE Runtime Environment (9.0+11) (build 9.0.1+11)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (9.0.1+11, mixed mode, tiered, compressed oops, g1 gc, windows-amd64)
# Problematic frame:
# C  [jSSC-2.8_x86_64.dll+0xb5db]
#
# No core dump will be written. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  S U M M A R Y ------------

Command Line: -Xmx256m UniversalGcodeSender.jar

Host: Intel(R) Core(TM) i7 CPU       M 620  @ 2.67GHz, 4 cores, 7G,  Windows 10 , 64 bit Build 16299 (10.0.16299.15)
Time: Sun Nov 26 17:15:42 2017 Central Europe Standard Time elapsed time: 4 seconds (0d 0h 0m 4s)

---------------  T H R E A D  ---------------

Current thread (0x0000020db8a7e800):  JavaThread "AWT-EventQueue-0" [_thread_in_native, id=2672, stack(0x0000004ed7200000,0x0000004ed7300000)]

Stack: [0x0000004ed7200000,0x0000004ed7300000],  sp=0x0000004ed72fae30,  free space=1003k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [jSSC-2.8_x86_64.dll+0xb5db]
C  [jSSC-2.8_x86_64.dll+0x12590]
C  [jSSC-2.8_x86_64.dll+0x1595]
C  0x0000020da4c1919c

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  jssc.SerialNativeInterface.openPort(Ljava/lang/String;Z)J+0
j  jssc.SerialPort.openPort()Z+65
j  com.willwinder.universalgcodesender.connection.JSSCConnection.openPort(Ljava/lang/String;I)Z+27
j  com.willwinder.universalgcodesender.AbstractCommunicator.openCommPort(Ljava/lang/String;I)Z+72
j  com.willwinder.universalgcodesender.BufferedCommunicator.openCommPort(Ljava/lang/String;I)Z+3
j  com.willwinder.universalgcodesender.AbstractController.openCommPort(Ljava/lang/String;I)Ljava/lang/Boolean;+27
j  com.willwinder.universalgcodesender.model.GUIBackend.openCommConnection(Ljava/lang/String;I)Z+8
j  com.willwinder.universalgcodesender.model.GUIBackend.connect(Ljava/lang/String;Ljava/lang/String;I)V+103
j  com.willwinder.universalgcodesender.MainWindow.opencloseButtonActionPerformed(Ljava/awt/event/ActionEvent;)V+78
j  com.willwinder.universalgcodesender.MainWindow.access$5400(Lcom/willwinder/universalgcodesender/MainWindow;Ljava/awt/event/ActionEvent;)V+2
j  com.willwinder.universalgcodesender.MainWindow$45.actionPerformed(Ljava/awt/event/ActionEvent;)V+5
j  javax.swing.AbstractButton.fireActionPerformed(Ljava/awt/event/ActionEvent;)V+84 java.desktop@9.0.1
j  javax.swing.AbstractButton$Handler.actionPerformed(Ljava/awt/event/ActionEvent;)V+5 java.desktop@9.0.1
j  javax.swing.DefaultButtonModel.fireActionPerformed(Ljava/awt/event/ActionEvent;)V+34 java.desktop@9.0.1
j  javax.swing.DefaultButtonModel.setPressed(Z)V+117 java.desktop@9.0.1
j  javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Ljava/awt/event/MouseEvent;)V+35 java.desktop@9.0.1
j  java.awt.Component.processMouseEvent(Ljava/awt/event/MouseEvent;)V+64 java.desktop@9.0.1
j  javax.swing.JComponent.processMouseEvent(Ljava/awt/event/MouseEvent;)V+23 java.desktop@9.0.1
j  java.awt.Component.processEvent(Ljava/awt/AWTEvent;)V+81 java.desktop@9.0.1
j  java.awt.Container.processEvent(Ljava/awt/AWTEvent;)V+18 java.desktop@9.0.1
j  java.awt.Component.dispatchEventImpl(Ljava/awt/AWTEvent;)V+496 java.desktop@9.0.1
j  java.awt.Container.dispatchEventImpl(Ljava/awt/AWTEvent;)V+42 java.desktop@9.0.1
j  java.awt.Component.dispatchEvent(Ljava/awt/AWTEvent;)V+2 java.desktop@9.0.1
j  java.awt.LightweightDispatcher.retargetMouseEvent(Ljava/awt/Component;ILjava/awt/event/MouseEvent;)V+327 java.desktop@9.0.1
j  java.awt.LightweightDispatcher.processMouseEvent(Ljava/awt/event/MouseEvent;)Z+155 java.desktop@9.0.1
j  java.awt.LightweightDispatcher.dispatchEvent(Ljava/awt/AWTEvent;)Z+50 java.desktop@9.0.1
j  java.awt.Container.dispatchEventImpl(Ljava/awt/AWTEvent;)V+12 java.desktop@9.0.1
j  java.awt.Window.dispatchEventImpl(Ljava/awt/AWTEvent;)V+19 java.desktop@9.0.1
j  java.awt.Component.dispatchEvent(Ljava/awt/AWTEvent;)V+2 java.desktop@9.0.1
j  java.awt.EventQueue.dispatchEventImpl(Ljava/awt/AWTEvent;Ljava/lang/Object;)V+41 java.desktop@9.0.1
j  java.awt.EventQueue.access$500(Ljava/awt/EventQueue;Ljava/awt/AWTEvent;Ljava/lang/Object;)V+3 java.desktop@9.0.1
j  java.awt.EventQueue$3.run()Ljava/lang/Void;+32 java.desktop@9.0.1
J 2673 c1 java.awt.EventQueue$3.run()Ljava/lang/Object; java.desktop@9.0.1 (5 bytes) @ 0x0000020da57088c4 [0x0000020da5708840+0x0000000000000084]
[... 482 lines cut ...]
Launcher Type: SUN_STANDARD

Logging:
Log output configuration:
#0: stdout all=warning uptime,level,tags
#1: stderr all=off uptime,level,tags

Environment Variables:
CLASSPATH=.;C:\Program Files (x86)\Java\jre7\lib\ext\QTJava.zip
PATH=C:\ProgramData\Oracle\Java\javapath;C:\Program Files\Common Files\Microsoft Shared\Windows Live;C:\Program Files (x86)\Common Files\Microsoft Shared\Windows Live;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\ThinkPad\Bluetooth Software\;C:\Program Files\ThinkPad\Bluetooth Software\syswow64;C:\Program Files (x86)\Common Files\Lenovo;C:\Program Files (x86)\Lenovo\Access Connections\;C:\Program Files (x86)\ [... cut ...]
OS: Windows 10 , 64 bit Build 16299 (10.0.16299.15)

CPU:total 4 (initial active 4) (2 cores per cpu, 2 threads per core) family 6 model 37 stepping 5, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, sse4.1, sse4.2, popcnt, aes, clmul, ht, tsc, tscinvbit

Memory: 4k page, physical 8244908k(4879388k free), swap 16633516k(13068828k free)

vm_info: Java HotSpot(TM) 64-Bit Server VM (9.0.1+11) for windows-amd64 JRE (9.0.1+11), built on Sep 28 2017 04:54:03 by "mach5one" with MS VC++ 12.0 (VS2013)

END.

russmufc, 2020-06-29:
Hi
I know this is an old thread but I'm having the same issue. UGS has been working fine (2.0 platform) for the past few months, then after a windows update (windows 10) it wont open at all. The start up screen appears and then just disappears when its half loaded. I've tried deleting/reinstalling but with no luck. Any ideas as to what I can try?

Thanks
Russ

JacksCrafts, 2020-06-29:
Did you update your Java?

itharper85, 2021-02-09:
> Hi
> I know this is an old thread but I'm having the same issue. UGS has been working fine (2.0 platform) for the past few months, then after a windows update (windows 10) it wont open at all. The [...]

I have uninstalled java, reinstalled the latest java, reinstalled the latest stable windows and all platform UGS as well as the nightly build and all of them begin to load and then the window "disappears" during a "Loading module services" loading phase. Do you have any recommendations.

breiler, 2021-02-09:
@itharper85 without a Windows version or [log files](https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files) it is difficult to give any advice. 

Here are some things you could check:
- We have seen users having problems running UGS together with their antivirus software, try disabling it.
- We have seen users having "special" characters in the folder path (Ie: c:\users\Göran\ugs\), try placing it in a folder path without special characters. This should technically not be a problem, but on some systems this doesn't work.
- If you are running a 32-bit windows you should run the latest nightly build. In previous versions we were bundling UGS with a 64-bit java which didn't on older machines. You should not need to install Java anymore if you are using the platform specific variants from here: [Downloads](https://github.com/winder/Universal-G-Code-Sender#downloads)
- Try removing previous settings from your home folder "c:\users\<username>\.ugsplatform" and restart

indipeers, 2023-03-10:
Hola buenas. Soy nuevo en el mundo del UGS y del CNC. He visto que varias personas han tenido también el problema del error [jSSC-2.8_x86_64.dll+0xb5db].
No entiendo muy bien porque me sale este error...les cuento un poco.
Mi antiguo ordenador murió(pobrecito mío, con lo bien que funcionaba) y compre un ordenador nuevo con el sistema operativo Windows 10 Home. Instale como siempre Arduino con su GCODE ( Arduino es lo que me enlaza el CNC con el ordenador), instale la versión de Java Para UGS y finalmente instale UGS. Lo raro es que cuando instale todo esto intente hacer una prueba y todo funciono genial.Pero al cabo de unos días volví para cortar unas maderas y cuando abrí UGS y le di ha Open(que es cundo se buelca el GCODE) automaticamente el programa de UGS se cierra y me sale un archivo de error. Es el siguiente archivo:

[hs_err_pid2832 (1).log](https://github.com/winder/Universal-G-Code-Sender/files/10941468/hs_err_pid2832.1.log)

No entiendo que la primera vez funcionara y luego sin hacer ningún cambio de nada , saliera este error. No se si Windows10 Home actualizó alguna cosa(como Java) y que por ese motivo ahora salga ese error...
Lo he intentado todo(dentro de mis conocimientos) : he formateado varias veces el ordenador y he reinstalado Windows, he probado varias versiones de UGS y de Java, he mirado de habilitar los Minidumps para Windows, pero el error no se soluciona... estoy un poco desesperado ya que llevo mas de 2meses intentando buscar una solución a esto pero ya no se que mas puedo hacer y estoy perdiendo algunos proyectos de trabajo importantes....
SI hay alguien que me pueda echar una mano para intentar solucionar esto problema se lo agradecería muchísimo. :)

(debajo les dejo lo que muestra el archivo de error)

#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=2832, tid=0x000000000000073c
#
# JRE version: Java(TM) SE Runtime Environment (8.0_281-b09) (build 1.8.0_281-b09)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.281-b09 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [jSSC-2.8_x86_64.dll+0xb5db]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  T H R E A D  ---------------

Current thread (0x000001c2011f5800):  JavaThread "AWT-EventQueue-0" [_thread_in_native, id=1852, stack(0x0000001451e00000,0x0000001451f00000)]

siginfo: ExceptionCode=0xc0000005, reading address 0x0000000001ee9fe2

Registers:
RAX=0x0000000001ee9fe2, RBX=0x0000001451efc960, RCX=0x00007ff8de9fcf74, RDX=0x0000000000000000
RSP=0x0000001451efc8f0, RBP=0x0000001451efca70, RSI=0x000001c2011f59f8, RDI=0x00000000000006f4
R8 =0x0000001451efc7e8, R9 =0x0000001451efc8e0, R10=0x0000000000000000, R11=0x0000000000000246
R12=0x0000001451efcb28, R13=0x000001c27f43cdd0, R14=0x0000001451efca20, R15=0x000001c2011f5800
RIP=0x000000007110b5db, EFLAGS=0x0000000000010202

Top of Stack: (sp=0x0000001451efc8f0)
0x0000001451efc8f0:   000001c27f34e100 00000006de9a06e1
0x0000001451efc900:   000000007110b230 000001c27f43cd20
0x0000001451efc910:   0000001451efca70 0000000000000004
0x0000001451efc920:   000001c2011f59f8 00000000000006f4
0x0000001451efc930:   0000001451efca70 0000000071112590
0x0000001451efc940:   000001c2011f5800 0000001451efca20
0x0000001451efc950:   000001c27f43cdd0 0000001451efcb28
0x0000001451efc960:   000001c27f43cdd0 00007ff8de9a06e1
0x0000001451efc970:   0000001451efc9a0 000001c259250000
0x0000001451efc980:   0000001451efca70 0000000000000000
0x0000001451efc990:   0000000071113500 0000000071113be8
0x0000001451efc9a0:   0000001451efc9d0 0000000071112641
0x0000001451efc9b0:   0000001451efc940 000001c2011f59f8
0x0000001451efc9c0:   0000000000000000 0000000000000000
0x0000001451efc9d0:   0000001451efca70 0000000071101595
0x0000001451efc9e0:   000000000000001c 000001c200000000 

Instructions: (pc=0x000000007110b5db)
0x000000007110b5bb:   c5 0f 1f 40 00 55 57 56 53 48 83 ec 28 48 8b 05
0x000000007110b5cb:   39 33 01 00 48 89 cb 48 85 c0 0f 84 c5 00 00 00
0x000000007110b5db:   8b 38 85 ff 0f 88 ab 00 00 00 48 8b 05 1c 33 01
0x000000007110b5eb:   00 48 85 c0 0f 84 bc 00 00 00 8b 30 85 f6 74 57 

Register to memory mapping:

RAX=0x0000000001ee9fe2 is an unknown value
[... 251 lines cut ...]

VM Arguments:
java_command: C:\Users\DELL\Desktop\uCCBViewer-2.5.jar
java_class_path (initial): C:\Users\DELL\Desktop\uCCBViewer-2.5.jar
Launcher Type: SUN_STANDARD

Environment Variables:
PATH=C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;c:\ediabas\bin;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\DTS\Binn\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\Users\DELL\AppData\Local\Microsoft\WindowsApps;
USERNAME=DELL
OS=Windows_NT
[... cut ...]
CPU:total 8 (initial active 8) (4 cores per cpu, 2 threads per core) family 6 model 142 stepping 12, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, sse4.1, sse4.2, popcnt, avx, avx2, aes, clmul, erms, 3dnowpref, lzcnt, ht, tsc, tscinvbit, bmi1, bmi2, adx

Memory: 4k page, physical 33360600k(28104056k free), swap 38603480k(31352564k free)

vm_info: Java HotSpot(TM) 64-Bit Server VM (25.281-b09) for windows-amd64 JRE (1.8.0_281-b09), built on Dec  9 2020 13:54:07 by "" with MS VC++ 15.9 (VS2017)

time: Tue Mar  9 11:10:17 2021
timezone: ?rodkowoeuropejski czas stand.
elapsed time: 3.008615 seconds (0d 0h 0m 3s)

breiler, 2023-03-10:
@indipeers please create a new issue

breiler, 2023-03-10:
@indipeers, wait what is C:\Users\DELL\Desktop\uCCBViewer-2.5.jar?
This is not a log from UGS?

indipeers, 2023-03-10:
> @indipeers, espera, ¿qué es C:\Users\DELL\Desktop\uCCBViewer-2.5.jar? ¿Esto no es un registro de UGS?

hola @breiler ! ahora mirare de crear un nuevo problema (perdona que yo no soy muy bueno con esto de los foros , ordenadores , informatica,etc)

Primero de todo no consigo ver el mensage que mande hace un momento...porque no se donde has visto lo de : 
C:\Users\DELL\Desktop\uCCBViewer-2.5.jar?
 Lo que si que te puedo decir es que DELL era el ordenador que tenia antes...y no recuero que UGS me pidiera ningún registro...no se..
@breiler me podrías decir como crear un nuevo enlace de error en el foro para poder hablar de esto?
Y muchas gracias por responder tan rapido!! :)

indipeers, 2023-03-10:
ya consegui ver mi mensage ....jajajaj perdon

breiler, 2023-03-10:
Go here to create a new bug report:
[image]

indipeers, 2023-03-10:
@breiler ya he visto donde sale lo de:
C:\Users\DELL\Desktop\uCCBViewer-2.5.jar?

vale @breiler voy a hacer que me salga de nuevo el archivo de error (porque creo que el que he mandado no es igual ya que no sale lo de " C:\Users\DELL\Desktop\uCCBViewer-2.5.jar " ).
Y así miro de crear un "nuevo problema" para poder comentarlo en el foro.
Gracias por todo @breiler