winder/Universal-G-Code-Sender #792: No core dump will be written. Minidumps are not enabled by default on client versions of Windows

sofoniyas, 2017-11-05:


winder, 2017-11-05:
Try a newer version of UGS, this one is very old and there have been many fixed and improvements since then.

closed 2017-11-05 by winder

sofoniyas, 2017-11-26:
tanks winder a lot, when i use UGS  V1.0.9 and connect the pc to the arduino the UGS V1.0.9 immediately closes

winder, 2017-11-26:
Does the nightly build do that as well?

sofoniyas, 2017-12-03:
yes, after setting the com ports when i press <open> it immediately closes and saves the error in Notepad extension. i attached it bellow if it is helping, i cant understand it . pls winder help me

[hs_err_pid5856.log](https://github.com/winder/Universal-G-Code-Sender/files/1524586/hs_err_pid5856.log)

winder, 2017-12-03:
I haven't tried running UGS with Java-9 at this point,maybe Java-8 will work.