winder/Universal-G-Code-Sender #794: Dowel Maker Plugin

winder, 2017-11-07:
I saw someone making dowels with their CNC recently. This seems like a good candidate for a UGS "Dowel Maker" plugin.

[image]

jahnj0584, 2017-11-07:
Conversational Gcode creation incoming????

winder, 2017-11-08:
@jahnj0584 maybe :)

[image]

jahnj0584, 2017-11-08:
Wicked. I'd be happy with one for rectangles, facing, and I guess this
counts as a circle?

winder, 2017-11-08:
@jahnj0584 all of those sound even simpler than this one. I think the floodgate is open now so I should be able to make few of these pretty quickly once I finish up this up.

vMeph, 2017-11-09:
@winder 
a cool plug in would be something like this
https://inkscape.org/pt/~Neon22/%E2%98%85lasercut-tabbed-box
would it be possible to do it
this wpuld be great to do boxes etc..

closed 2017-11-30 by winder