winder/Universal-G-Code-Sender #795: Unable to open visualizer: VisualizerTopComponent could not be located or loaded from Components folder

oddish2211, 2017-11-08:
I just checked out the latest nightly and when opening the visualizer i get the following exception.
I've pasted a bit from messages.log, if you need anytrhing more just ask.

`INFO [org.netbeans.core.windows.persistence]: [PersistenceManager.getTopComponentForID] Problem when deserializing TopComponent for tcID:'VisualizerTopComponent'. Reason: Top component VisualizerTopComponent could not be located or loaded from Components folder.
INFO [org.netbeans.core.startup.layers.BinaryFS]: value = com.willwinder.ugs.nbm.visualizer.Visualizer2TopComponent from Actions/Window/com-willwinder-ugs-nbm-visualizer-Visualizer2TopComponent.instance
java.lang.InternalError: XXX0 profile[1]: GL3bc -> profileImpl GL4bc !!! not mapped 
	at com.jogamp.opengl.GLProfile.computeProfileMap(GLProfile.java:2071)
	at com.jogamp.opengl.GLProfile.initProfilesForDeviceCritical(GLProfile.java:1954)
	at com.jogamp.opengl.GLProfile.initProfilesForDevice(GLProfile.java:1875)
	at com.jogamp.opengl.GLProfile.initProfilesForDefaultDevices(GLProfile.java:1842)
	at com.jogamp.opengl.GLProfile.access$000(GLProfile.java:80)
	at com.jogamp.opengl.GLProfile$1.run(GLProfile.java:230)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.jogamp.opengl.GLProfile.initSingleton(GLProfile.java:216)
	at com.jogamp.opengl.GLProfile.getDefaultDevice(GLProfile.java:2027)
	at com.jogamp.opengl.GLCapabilities.<init>(GLCapabilities.java:84)
	at com.willwinder.ugs.nbm.visualizer.Visualizer2TopComponent.<init>(Visualizer2TopComponent.java:71)
Caused: java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
[catch] at org.netbeans.core.startup.layers.BinaryFS$AttrImpl.getValue(BinaryFS.java:712)
	at org.netbeans.core.startup.layers.BinaryFS$BFSBase.getAttribute(BinaryFS.java:549)
	at org.openide.filesystems.MultiFileObject.getAttribute(MultiFileObject.java:944)
	at org.openide.filesystems.MultiFileObject.getAttribute(MultiFileObject.java:877)
	at org.openide.filesystems.MultiFileObject.getAttribute(MultiFileObject.java:940)
	at org.openide.filesystems.MultiFileObject.getAttribute(MultiFileObject.java:877)
	at org.openide.filesystems.MultiFileObject.getAttribute(MultiFileObject.java:798)
	at org.netbeans.core.startup.layers.BinaryFS$FileMap.get(BinaryFS.java:1128)
	at org.openide.windows.OpenComponentAction.getTopComponent(OpenComponentAction.java:79)
	at org.openide.windows.OpenComponentAction.actionPerformed(OpenComponentAction.java:89)
	at org.openide.awt.AlwaysEnabledAction$1.run(AlwaysEnabledAction.java:199)
	at org.openide.util.actions.ActionInvoker$1.run(ActionInvoker.java:95)
	at org.openide.util.actions.ActionInvoker.doPerformAction(ActionInvoker.java:116)
	at org.openide.util.actions.ActionInvoker.invokeAction(ActionInvoker.java:99)
	at org.openide.awt.AlwaysEnabledAction.actionPerformed(AlwaysEnabledAction.java:202)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:877)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at org.netbeans.core.TimableEventQueue.dispatchEvent(TimableEventQueue.java:159)
[... 31 lines cut ...]
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at org.netbeans.core.TimableEventQueue.dispatchEvent(TimableEventQueue.java:159)
[catch] at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding control state listener.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding controller listener.
INFO [com.willwinder.universalgcodesender.utils.SettingsFactory]: Saving settings.`

winder, 2017-11-08:
This error is coming out of the OpenGL native library used by UGS. So your best bet is to make sure you're graphic card has an up to date driver.

closed 2017-11-17 by winder

oddish2211, 2017-11-19:
Well i share your opinion that problem could be in the graphics driver. Unfortunately i already have the latest version of amdgpu/mesa available for Debian Buster.

A workaround till mesa is fixed is acting like i only support version 1.30 of GLSL using the following environment variable when starting run_platform.sh:
`MESA_GLSL_VERSION_OVERRIDE=130`

Still thanks for showing me the direction.

winder, 2017-11-19:
Thanks for the update