winder/Universal-G-Code-Sender #798: UGCS 1.0.9 wrong tag?

nachoparga, 2017-11-11:
Hi

I'm trying to recompile ugcs tag 1.0.9

https://github.com/winder/Universal-G-Code-Sender/archive/v1.0.9.zip

However when i run mvn install  the log states that is UGS2.0
Even wen i run the target generated  it stated that is version 2 and tht is a nightly build

Is the zip mentioned wrong?
Am i doing something wrong?

INFO] Scanning for projects...
[INFO] 
[INFO] ------------------------------------------------------------------------
[INFO] Building UGS 2.0
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- maven-dependency-plugin:2.10:unpack (unpack) @ UGS ---
[INFO] Configured Artifact: org.jogamp.gluegen:gluegen-rt:natives-android-armv6:2.3.1:jar
[INFO] Configured Artifact: org.jogamp.gluegen:gluegen-rt:natives-linux-amd64:2.3.1:jar
[INFO] Configured Artifact: org.jogamp.gluegen:gluegen-rt:natives-linux-armv6:2.3.1:jar
[INFO] Configured Artifact: org.jogamp.gluegen:gluegen-rt:natives-linux-armv6hf:2.3.1:jar
[INFO] Configured Artifact: org.jogamp.gluegen:gluegen-rt:natives-linux-i586:2.3.1:jar
[INFO] Configured Artifact: org.jogamp.gluegen:gluegen-rt:natives-macosx-universal:2.3.1:jar
[INFO] Configured Artifact: org.jogamp.gluegen:gluegen-rt:natives-windows-amd64:2.3.1:jar
[INFO] Configured Artifact: org.jogamp.gluegen:gluegen-rt:natives-windows-i586:2.3.1:jar
[INFO] Configured Artifact: org.jogamp.gluegen:gluegen-rt:natives-solaris-amd64:2.3.1:jar
[INFO] Configured Artifact: org.jogamp.gluegen:gluegen-rt:natives-solaris-i586:2.3.1:jar
[INFO] Configured Artifact: org.jogamp.jogl:jogl-all:natives-android-armv6:2.3.1:jar
[INFO] Configured Artifact: org.jogamp.jogl:jogl-all:natives-linux-amd64:2.3.1:jar
[INFO] Configured Artifact: org.jogamp.jogl:jogl-all:natives-linux-armv6:2.3.1:jar
[INFO] Configured Artifact: org.jogamp.jogl:jogl-all:natives-linux-armv6hf:2.3.1:jar
[INFO] Configured Artifact: org.jogamp.jogl:jogl-all:natives-linux-i586:2.3.1:jar
[INFO] Configured Artifact: org.jogamp.jogl:jogl-all:natives-macosx-universal:2.3.1:jar
[INFO] Configured Artifact: org.jogamp.jogl:jogl-all:natives-windows-amd64:2.3.1:jar
[INFO] Configured Artifact: org.jogamp.jogl:jogl-all:natives-windows-i586:2.3.1:jar
[INFO] Configured Artifact: org.jogamp.jogl:jogl-all:natives-solaris-amd64:2.3.1:jar
[INFO] Configured Artifact: org.jogamp.jogl:jogl-all:natives-solaris-i586:2.3.1:jar
[INFO] gluegen-rt-2.3.1-natives-android-armv6.jar already unpacked.
[INFO] gluegen-rt-2.3.1-natives-linux-amd64.jar already unpacked.
[INFO] gluegen-rt-2.3.1-natives-linux-armv6.jar already unpacked.
[INFO] gluegen-rt-2.3.1-natives-linux-armv6hf.jar already unpacked.
[INFO] gluegen-rt-2.3.1-natives-linux-i586.jar already unpacked.
[INFO] gluegen-rt-2.3.1-natives-macosx-universal.jar already unpacked.
[INFO] gluegen-rt-2.3.1-natives-windows-amd64.jar already unpacked.
[INFO] gluegen-rt-2.3.1-natives-windows-i586.jar already unpacked.
[INFO] gluegen-rt-2.3.1-natives-solaris-amd64.jar already unpacked.
[INFO] gluegen-rt-2.3.1-natives-solaris-i586.jar already unpacked.
[INFO] jogl-all-2.3.1-natives-android-armv6.jar already unpacked.
[INFO] jogl-all-2.3.1-natives-linux-amd64.jar already unpacked.
[INFO] jogl-all-2.3.1-natives-linux-armv6.jar already unpacked.
[INFO] jogl-all-2.3.1-natives-linux-armv6hf.jar already unpacked.
[INFO] jogl-all-2.3.1-natives-linux-i586.jar already unpacked.
[INFO] jogl-all-2.3.1-natives-macosx-universal.jar already unpacked.
[INFO] jogl-all-2.3.1-natives-windows-amd64.jar already unpacked.
[INFO] jogl-all-2.3.1-natives-windows-i586.jar already unpacked.
[INFO] jogl-all-2.3.1-natives-solaris-amd64.jar already unpacked.
[INFO] jogl-all-2.3.1-natives-solaris-i586.jar already unpacked.
[INFO] 
[INFO] --- maven-resources-plugin:2.6:resources (default-resources) @ UGS ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] Copying 30 resources
[INFO] Copying 1 resource
[INFO] 
[INFO] --- maven-compiler-plugin:3.1:compile (default-compile) @ UGS ---
[INFO] Nothing to compile - all classes are up to date
[INFO] 
[INFO] --- maven-resources-plugin:2.6:testResources (default-testResources) @ UGS ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[... 40 lines cut ...]
[INFO] Including javax.media:jmf:jar:2.1.1e in the shaded jar.
[INFO] Including org.swinglabs:swing-layout:jar:1.0.3 in the shaded jar.
[INFO] Including org.eclipse.jetty:jetty-server:jar:8.1.16.v20140903 in the shaded jar.
[INFO] Including org.eclipse.jetty.orbit:javax.servlet:jar:3.0.0.v201112011016 in the shaded jar.
[INFO] Including org.eclipse.jetty:jetty-continuation:jar:8.1.16.v20140903 in the shaded jar.
[INFO] Including org.eclipse.jetty:jetty-http:jar:8.1.16.v20140903 in the shaded jar.
[INFO] Including org.eclipse.jetty:jetty-io:jar:8.1.16.v20140903 in the shaded jar.
[INFO] Including org.eclipse.jetty:jetty-util:jar:8.1.16.v20140903 in the shaded jar.
[INFO] Including com.google.code.gson:gson:jar:2.2.4 in the shaded jar.
[INFO] Including org.apache.commons:commons-lang3:jar:3.3.1 in the shaded jar.
[INFO] Including org.scream3r:jssc:jar:2.8.0 in the shaded jar.
[INFO] Including com.google.zxing:javase:jar:3.2.0 in the shaded jar.
[INFO] Including com.google.zxing:core:jar:3.2.0 in the shaded jar.
[INFO] Replacing c:\Eclipse\Universal-G-Code-Sender-1.0.9\Universal-G-Code-Sender-1.0.9\target\UniversalGcodeSender.jar with c:\Eclipse\Universal-G-Code-Sender-1.0.9\Universal-G-Code-Sender-1.0.9\target\UGS-2.0-shaded.jar
[INFO] 
[INFO] --- maven-install-plugin:2.4:install (default-install) @ UGS ---
[INFO] Installing c:\Eclipse\Universal-G-Code-Sender-1.0.9\Universal-G-Code-Sender-1.0.9\target\UGS-2.0.jar to C:\Users\Casa\.m2\repository\com\willwinder\universalgcodesender\UGS\2.0\UGS-2.0.jar
[INFO] Installing c:\Eclipse\Universal-G-Code-Sender-1.0.9\Universal-G-Code-Sender-1.0.9\pom.xml to C:\Users\Casa\.m2\repository\com\willwinder\universalgcodesender\UGS\2.0\UGS-2.0.pom
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 6.606 s
[INFO] Finished at: 2017-11-11T20:45:06+01:00
[INFO] Final Memory: 17M/269M
[INFO] ------------------------------------------------------------------------

winder, 2017-11-12:
Check out `com.willwinder.universalgcodesender.utils.Version` for the version string, I haven't really used maven for versioning

winder, 2017-11-12:
It's also possible that I changed the version locally to 1.0.9 when I made the release.

closed 2017-11-12 by winder

nachoparga, 2017-11-12:
private static final String VERSION = "2.0 [nightly] ";
So my question is how do i download 1.0.9 sources?
As i said this is the link used for downloading it ...
https://github.com/winder/Universal-G-Code-Sender/archive/v1.0.9.zip

winder, 2017-11-14:
@nachoparga I probably changed the version string manually to build the 1.0.9 release. That is the right code.