winder/Universal-G-Code-Sender #799: Weird issue with UGS and arduino

jahnj0584, 2017-11-12:
Take a look 

https://youtu.be/sV1QisRismk

As SOON as I turn the spindle on (VFD itself is fine) the arduino/UGS lock up and have a massive delay in sending the commands, as well as even receiving them. The M5 command should be instantly sent to the console when I hit enter and the spindle shuts off. Instead, there's a 4 second delay to send the M5,and about 8 for the spindle to shut off. 

Only cure is to close UGS and hit the reset button on the arduino. 

It was working perfectly fine. Only change from last week is I put windows 8 on the computer but why would that dictate any of this.... Programs send find until spindle is enabled.

winder, 2017-11-12:
The video showed that GRBL was no longer responding, so my guess is some sort of EMI is causing problems with the USB/Arduino.

jahnj0584, 2017-11-13:
Just wish I knew why the EMI only became an issue after my windows 8
installation. I had everything running fine before...

aforww, 2017-11-13:
Trust me, it's noise. I just went through all of this with mine. How far is your VFD from the board? Mine (quality VFD not Chinese bundle VFD) was about 14 inches away and caused be all kinds of issues. Once I put it in the same spot but in a metal enclosure, never had a freeze again. 

Side question,  how are you getting those commands to start/stop? My Arduino based machine with latest GRBL and UGS will only accept M4 to start M3 to stop.

jahnj0584, 2017-11-13:
Well what metal box could I put it in today? Lol
I could move the VFD over another foot or two, I'll try that.

Going to try a different rpm today. It may just be a certain frequency that
causes issues.

I didn't do anything for commands. Make sure you don't have any expression
removers on.

aforww, 2017-11-13:
VFDs are mystical things lol. I had to get educated by my father who has a
lifetime installing them. Electrical noise is consistent sometimes, absent
others. Could be just a matter of the position of some wires, maybe your
USB cord.

I fabricated my own metal box out of some 18 gauge sheet metal. Here's a
link to my build if you're interested. I'd highly recommend shielding any
signal wiring within a few feet of the thing and the spindle power cord.

winder, 2017-11-14:
@jahnj0584 you could use a couple sheets of tin foil to test things out

closed 2017-11-14 by winder