winder/Universal-G-Code-Sender #801: Preview screen is does not include all the drawing if it has negative values in the g-code

FARose, 2017-11-14:
The display will not show the left part of the drawing unless I rotate the image.  It will not center correctly and show the whole thing.

Also, when changing the units to inches, the machine status stays in mm.

I like this new 2.0 layout much more than the 1.09 version.  I have used MACH3 and like some of the features it has and suspect more MACH3 uses would like a similar set of features.

I am using this for a CNC machine and not a 3D printer.

winder, 2017-11-16:
@FARose do you have an example that you could share? This file works fine and has the origin in the center of the file: https://github.com/winder/Universal-G-Code-Sender/blob/master/test_files/shapeoko.txt

FARose, 2017-11-16:
> master/test_files/shapeoko.txt

Here is the file I had the issue with.  It was made to make three hot wire
cut foam bearing blocks.  There was dual stepper motors on the Z axes, one
on each end of the wire.
I have to figure out how to do that with the UGS software.

I installed the "Platform" version and it displayed alright on it, but not
on the non-platform version.

Thank you,
Fred

G20
G1
G1 X-2.5 Y2.2495 Z-2.5 A2.2495
G1 X-1.3625 Y2.2495 Z-1.3625 A2.2495
G1 X-1.3625 Y1.8745 Z-1.3625 A1.8745
G1 X-0.5625 Y1.8745 Z-0.5625 A1.8745
G1 X-0.5625 Y0.7884 Z-0.5625 A0.7884
G1 X-0.6175 Y0.7464 Z-0.6175 A0.7464
G1 X-0.6632 Y0.7062 Z-0.6632 A0.7062
G1 X-0.7062 Y0.6632 Z-0.7062 A0.6632
G1 X-0.7464 Y0.6175 Z-0.7464 A0.6175
G1 X-0.7837 Y0.5694 Z-0.7837 A0.5694
G1 X-0.8179 Y0.5191 Z-0.8179 A0.5191
G1 X-0.8489 Y0.4667 Z-0.8489 A0.4667
G1 X-0.8766 Y0.4125 Z-0.8766 A0.4125
G1 X-0.9007 Y0.3566 Z-0.9007 A0.3566
G1 X-0.9213 Y0.2994 Z-0.9213 A0.2994
G1 X-0.9383 Y0.2409 Z-0.9383 A0.2409
G1 X-0.9516 Y0.1815 Z-0.9516 A0.1815
G1 X-0.9611 Y0.1214 Z-0.9611 A0.1214
G1 X-0.9638 Y0.0931 Z-0.9638 A0.0931
G1 X-2.2294 Y-0.6376 Z-2.2294 A-0.6376
G1 X-1.6669 Y-1.6119 Z-1.6669 A-1.6119
G1 X-0.4015 Y-0.8813 Z-0.4015 A-0.8813
G1 X-0.3566 Y-0.9007 Z-0.3566 A-0.9007
G1 X-0.2994 Y-0.9213 Z-0.2994 A-0.9213
G1 X-0.2409 Y-0.9383 Z-0.2409 A-0.9383
G1 X-0.1875 Y-0.9503 Z-0.1875 A-0.9503
G1 X-0.1875 Y-0.9183 Z-0.1875 A-0.9183
G1 X-0.1875 Y-0.5633 Z-0.1875 A-0.5633
G1 X-0.2186 Y-0.5521 Z-0.2186 A-0.5521
G1 X-0.2528 Y-0.5373 Z-0.2528 A-0.5373
G1 X-0.2861 Y-0.5203 Z-0.2861 A-0.5203
G1 X-0.3182 Y-0.5014 Z-0.3182 A-0.5014
G1 X-0.349 Y-0.4804 Z-0.349 A-0.4804
G1 X-0.3785 Y-0.4575 Z-0.3785 A-0.4575
G1 X-0.3939 Y-0.4439 Z-0.3939 A-0.4439
G1 X-1.5296 Y-1.0996 Z-1.5296 A-1.0996
G1 X-1.7171 Y-0.7749 Z-1.7171 A-0.7749
G1 X-0.5833 Y-0.1113 Z-0.5833 A-0.1113
G1 X-0.5815 Y-0.1192 Z-0.5815 A-0.1192
G1 X-0.5891 Y-0.0744 Z-0.5891 A-0.0744
G1 X-0.5926 Y-0.0373 Z-0.5926 A-0.0373
G1 X-0.5938 Y0 Z-0.5938 A0
G1 X-0.5926 Y0.0373 Z-0.5926 A0.0373
G1 X-0.5891 Y0.0744 Z-0.5891 A0.0744
G1 X-0.5833 Y0.1113 Z-0.5833 A0.1113
G1 X-0.5751 Y0.1477 Z-0.5751 A0.1477
G1 X-0.5647 Y0.1835 Z-0.5647 A0.1835
G1 X-0.5521 Y0.2186 Z-0.5521 A0.2186
G1 X-0.5373 Y0.2528 Z-0.5373 A0.2528
G1 X-0.5203 Y0.2861 Z-0.5203 A0.2861
G1 X-0.5014 Y0.3182 Z-0.5014 A0.3182
G1 X-0.4804 Y0.349 Z-0.4804 A0.349
G1 X-0.4575 Y0.3785 Z-0.4575 A0.3785
G1 X-0.4329 Y0.4065 Z-0.4329 A0.4065
G1 X-0.4065 Y0.4329 Z-0.4065 A0.4329
G1 X-0.3785 Y0.4575 Z-0.3785 A0.4575
G1 X-0.349 Y0.4804 Z-0.349 A0.4804
[... 298 lines cut ...]
G1 X3.8566 Y0.9007 Z3.8566 A0.9007
G1 X3.9015 Y0.8813 Z3.9015 A0.8813
G1 X5.1669 Y1.6119 Z5.1669 A1.6119
G1 X5.7294 Y0.6376 Z5.7294 A0.6376
G1 X4.4638 Y-0.0931 Z4.4638 A-0.0931
G1 X4.4611 Y-0.1214 Z4.4611 A-0.1214
G1 X4.4516 Y-0.1815 Z4.4516 A-0.1815
G1 X4.4383 Y-0.2409 Z4.4383 A-0.2409
G1 X4.4213 Y-0.2994 Z4.4213 A-0.2994
G1 X4.4007 Y-0.3566 Z4.4007 A-0.3566
G1 X4.3766 Y-0.4125 Z4.3766 A-0.4125
G1 X4.3489 Y-0.4667 Z4.3489 A-0.4667
G1 X4.3179 Y-0.5191 Z4.3179 A-0.5191
G1 X4.2837 Y-0.5694 Z4.2837 A-0.5694
G1 X4.2464 Y-0.6175 Z4.2464 A-0.6175
G1 X4.2062 Y-0.6632 Z4.2062 A-0.6632
G1 X4.1632 Y-0.7062 Z4.1632 A-0.7062
G1 X4.1175 Y-0.7464 Z4.1175 A-0.7464
G1 X4.0625 Y-0.7884 Z4.0625 A-0.7884
G1 X4.0625 Y-1.8745 Z4.0625 A-1.8745
G1 X4.8625 Y-1.8745 Z4.8625 A-1.8745
G1 X4.8625 Y-2.2495 Z4.8625 A-2.2495
G1 X9.5 Y-2.2495 Z9.5 A-2.2495
G1 X9.5 Y2.2495 Z9.5 A2.2495
G1 X-2.5 Y2.2495 Z-2.5 A2.2495

winder, 2017-11-16:
@FARose do you have an old version of UGS? It seems to display properly on the classic and platform visualizers.

FARose, 2017-11-16:
I have two downloads of the nightly version 1.09 and I just tried each of
them.  One is not working correctly when I try to use the visualizer, but
the other does.  I have the 2.0 version which works fine also.

I must have had a problem when it downloaded and I will just delete that
one.

Thank you very much for looking into this!

I like layout of the platform version and will continue to learn how to use
it.  Is there a way to define two steppers, one a slave and have it not be
displayed...maybe i should have called it the A-axes and not the Z-axes.

Thanks again,
Fred

winder, 2017-11-16:
Great! UGS only supports XYZ right now, if you want to simulate a rotary axis you'll have to make sure your code maps the A axis to XY or Z.

closed 2017-11-16 by winder