winder/Universal-G-Code-Sender #802: UGS Platform on Rpi B+

khturnings, 2017-11-14:
First - Will this actually run in a Pi Version 1 B+ - 512meg one?
Second - I have install Raspbian Stretch and updated it.
In the terminal windows I see this this error - libEGL warning: DRI2: failed to authenticate but the tool opens, slowly :)
Thirdly - I loaded a simple GCode file and tried to visualise it. This error then appears - A com.jogamp.opengl.GLException exception has occurred.

Any help appreciated.

Can you add some information to the user guide regarding required specifications and setup.
I found on my laptop a Lenovo G570 has a Intel HD 3000 graphics card - OpenGL is not supported so it did not work on that platform, even though I found that Linux might support it (My head hurts)

Thanks in advance.

winder, 2017-11-14:
Unfortunately, the visualizer does not work for the RPi. See issue #47 

It should be possible to get it working because JOGL supports RPi, but nobody has provided a PR to fix it and I've been focused on other things.

closed 2017-11-14 by winder

khturnings, 2017-11-14:
Thanks for your very speedy help - keep up the good work