winder/Universal-G-Code-Sender #805: State Panel picks up feedrate from jogging

carneeki, 2017-11-17:
The state panel picks up the feed rate when issuing a jog ($J), but the jog feed rate shouldn't be affecting G1 mode. When I issue a $G to check, the State panel corrects itself. Maybe the parser should ignore feed rates in jogging?

Similarly (in terms of parsing and displaying the results), do we want a few extras?
* Feed rate mode (units per minute [G94] / inverse time [G93])
* Spindle state (CW [M3] / CCW [M4] / OFF [M5])
* Coolant state (mist [M7] / flood [M8] / off [M9]) to be in the mix somewhere too? (possibly checkboxes as flood and mist are not mutually exclusive to each other). By default grbl ships with M8 only, but is trivial to enable secondary coolant.

winder, 2017-11-17:
Nice catche re: jogging.

Feed rate mode: I actually implemented this then commented out because the text made the minimum width wider than any of the other left side tabs. I need to find a solution, maybe just set a max width on the combo boxes.

Spindle/Coolant State: Good idea, same with the Tool number. I'm considering anything [listed in this table](http://linuxcnc.org/docs/html/gcode/overview.html#_modal_groups).

I was considering checkboxes for the grbl v1.1 endstops as well.

carneeki, 2017-11-17:
Tool numbering and a tool table in UGS would be pretty sweet. Maybe I'll figure out Maven and NetBeans this weekend to work on it.

breiler, 2019-03-14:
Regarding the feed speed from jogging, seems to not be a problem anymore. I took the freedom to change the "bug" tag with "future enhancement"